mne-tools/mne-python #10034: Access the current channel name within apply_function()

emma-bailey, 2021-11-24:
When writing custom functions to apply to MNE raw structures, it is often useful to be able to access the current channel that is being worked on - something which is currently not possible with apply_function() when working on multiple channels, where parallelisation is most useful.

Essentially an update to allow access to the channel this method is currently working on when using the njobs parameter to implement parallelization would be very useful!

hoechenberger, 2021-11-24:
Thanks for opening this feature request, @emma-bailey 

Just for reference, this came up on the forum, https://mne.discourse.group/t/determining-which-channel-apply-function-is-currently-working-on/4011

It's common in split-apply-combine situations that there's a way to retrieve some sort of index or description of the currently-processed chunk. So I believe we should definitely have this.

For epochs, we'd also need the epochs index.

agramfort, 2021-11-24:
can you use the forum to ask usage questions please?

also what do you want to do mathematically? Maybe there is a different
route...

>

emma-bailey, 2021-11-24:
I'm applying PCA OBS channel by channel to the data stored in an MNE raw structure - it's essentially an implementation of the algorithm described by Niazi et. al. https://pubmed.ncbi.nlm.nih.gov/16150610/ which is implemented in the fmrib toolbox for MATLAB but I haven't found anything similar in python/MNE so I've been writing my own - it seems to work but as it is designed to act channel by channel, and when fitting the artefact for removal, I want to save the start and end points of the fitting window utilised around each detected R-peak, but as this is channel specific information it would be convenient to have access to the channel number!

That's my specific use case, but I'm sure there are other applications where it would be handy to be able to access it :)

agramfort, 2021-11-24:
ok so it's a temporal PCA. you regress out some confounds from each time
course if I get it right

did you consider passing the picks param in apply_function? doing this the
function you apply
can be informed about the picks you pass

>

emma-bailey, 2021-11-24:
Yeah so currently I am passing in the picks param of a single channel, which means I have access to the channel because there's only one, the problem arises when I want to run this simultaneously on multiple channels and lose access to this level of control as I can't tell within my function which channel is being acted on :)

agramfort, 2021-11-24:
if you pass multiple picks and you do channel_wise=False then you get all
channels in your function
and you can do what you want?

>

emma-bailey, 2021-11-24:
My understanding was that channel_wise=False applies my function to all channels at the same time, but I wasn't aware it would allow me to access each channels name it is working on?

agramfort, 2021-11-24:
how about something like this:

```
import numpy as np
import mne
from mne.io import pick

data = np.random.randn(5, 100)
info = mne.create_info(list('abcde'), sfreq=100, ch_types='eeg')
raw = mne.io.RawArray(data, info)
print(raw.get_data().std(axis=1))
picks = [0, 2, 3]

def func(data):
    assert data.shape[0] == len(picks)
    for d, p in zip(data, picks):
        d[:] *= p
    return data

raw.apply_function(func, picks=picks, channel_wise=False)
print(raw.get_data().std(axis=1))
```

emma-bailey, 2021-11-24:
Hmmmm interesting! I was currently just going to use joblib to parallelise outside the function call but this might be a more elegant solution thanks!

agramfort, 2021-11-24:
great I could help. Feel free to reopen if necessary

hoechenberger, 2021-11-24:
Hello @agramfort,

like said above, this question actually was posted on the forum first; I then adviced @emma-bailey to open a feature request. In [my response](https://mne.discourse.group/t/determining-which-channel-apply-function-is-currently-working-on/4011/2?u=richard) I also suggested a workaround akin to what you proposed above: going parallel inside the `func` that `apply_func` calls.

However this **is** obviously a hack, as it disregards `apply_func`'s `n_jobs` parameter.

I therefore think we should keep this feature request open and discuss whether this is an improvement we want to implement. It's not a simple usage question. Therefore, re-opening for now and happy to discuss!

larsoner, 2022-02-21:
We could pretty easily use `_get_args` magic inspection of the `callable` to say: if `ch_name` is in the list of arguments of the callable, then we should pass `ch_name=ch_name` as well. This would be backward compatible and require no new parameters

dominikwelke, 2023-11-13:
hi i'd like to work on this issue

@larsoner - could you explain what you mean with the `_get_args` magic? i did a quick search but cant find much on it..

larsoner, 2023-11-13:
We no longer use _get_args, nowadays you can use inspect.getfullargspec

PR #12206 (2024-02-06): ENH: make apply_function aware of channel index
Fixes #10034.

this PR makes the index of currently processed channel available within custom functions applied to data.
this is needed for processing channels differently during parallel processing, as requested in #10034 .
closed 2024-02-06 by larsoner