mne-tools/mne-python #10053: Function "read_raw_edf"  have some channels which dosen't apear in version 0.23.0

0220220313, 2021-11-29:
I used args "exclude"  to get the specific channel data before version 0.23.0.
Since it's much faster than read all data and pick channel data.
But when I upgrade mne to 0.24.0 , it would have bug .
There have some channels that dosent apear in version 0.23.0.
It will appear this warning , and I cannot read them in raw edf data. 

RuntimeWarning: Channel names are not unique, found duplicates for: {'Impedance', 'Rate', 'Sum', 'Pressure', 'Fast', '(Imp)', 'Leg'}. Applying running numbers for duplicates.

And this is my function  to get specific channel data in 0.23.0.

```
def get_channel_name(edfname): ## Read  edf channel name
    with open(edfname,'r') as f:
        _ = f.read(96+120+36)
        channel_num = int(f.read(4))
        return  [f.read(16).strip() for i in range(channel_num )]

def get_channel_data(edfname,need_channels):
    ch_names = get_channel_name(edfname)
    exclude_channels = ['' if the_ch in need_channels else the_ch  for the_ch in ch_names ]
    data = mne.io.read_raw_edf(edfname,exclude=channel_list,verbose=False)
    return data.get_data() , data.info['sfreq'] , data.ch_names
```
Function "get_channel_name" is copy from function "_read_edf_header" in version 0.23.0.

cbrnr, 2021-11-29:
I would not recommend copying and using internal code with all these magic numbers (`96+120+36`), this is very brittle and likely to break anytime. In addition, you are opening the file in text mode instead of binary (should be `'rb'`).

In any case, this still works for me in 0.24. However, a better approach is to first load the data with `preload=False`. This won't load the data, but you will get the header information, including channel names. You can then use `raw["ch_names"]` to

- either construct the `exclude` argument and then load the data again with `preload=True` or
- drop channels you do not need with `raw.drop_channels()` and then load the data with `raw.load_data()`.

0220220313, 2021-11-29:
@cbrnr  Thanks for reply this issue.
I would try to use new method you provide to get the specfic channels data.
Because of the channels I need have different frequence so I use exclude .
And I found out why I received a warning about duplicate channel names.
In function "_read_edf_header" in v0.24.0
```
        ch_names = []
        ch_types = []
        for ch_idx, this_label in enumerate(ch_labels):
            # if no channel type, then we will default to eeg
            ch_type = None
            ch_name = this_label

            # space is found, so the prefix is the channel type. 'Annotations'
            # is also a keyword we search for
            if (this_label.count(' ') == 1) and \
                    ('Annotations' not in this_label):
                ch_type, ch_name = this_label.split(' ')

                # channel types should be upper case for easy comparison
                ch_type = ch_type.upper()
```
It would split channels name by space and cause some channels have the same name.
Maybe the EDF files I owned is not in typical format....
My channel names contain a lot of spaces.
Anyway, thank you for answering my question.

cbrnr, 2021-11-29:
Glad I could help and that you could solve your issue! EDF channel names are poorly specified, and you are correct that from 0.24 we now detect if they contain spaces. We use this information to deduce channel types, which works for common cases such as `"EEG Fz"`. Apparently, this should only be the case if we detect [known prefixes](https://mne.tools/stable/generated/mne.io.read_raw_edf.html#:~:text=for%20EDF%20files.-,Notes,-It%20is%20worth).

Could you please post the channel names you have in (one of) your files? If these are unique, the warning about duplicate names should not appear, but if it does this might indicate that we're not doing the correct thing. Thanks!

0220220313, 2021-11-29:
My channels contains :
E1 ,
E1 Impedance,
O1 ,
O1 Impedance

Then the channles "Impedance" would be duplicated. 

But I think this is my own file format problem.
I even need to write another header reader to solve this problem because It have some channel names longer than 16.

cbrnr, 2021-11-29:
Thanks @0220220313! I think at least partly MNE-Python is doing the wrong thing here, there shouldn't be any duplicate channels names because these are unknown prefixes. Could you share one of your files so that I can reproduce the problem? If you have some channel names that are too long this would be a separate issue.

0220220313, 2021-11-30:
I am glad to help you.
Here is the example EDF .
[example.zip](https://github.com/mne-tools/mne-python/files/7622205/example.zip)

Some of the origin channel names in UI of my device are longer then 16 , they would be wrote only 16  chars in edf.
Since the final char of channel name would be drop by function "_read_edf_header " , if the channel name lenth is 16 chars.
I need to wite another function to read channel names.
Therefore, I think there is something wrong with my device when exporting EDF files.

cbrnr, 2021-11-30:
Thanks! You are correct that EDF channel labels are 16 characters at most. If you have longer labels originally, then yes, I think you should try to shorten the names if possible so that nothing is lost when you export to EDF.

However, the first few channel labels in your example file are:
```
1
1 Impedance
1-2
1-2 (Imp)
1-F
2
2 Impedance
```

MNE-Python reads these as:
```
1
Impedance-0
1-2
(Imp)-0
1-F
2
Impedance-1
```

So even though unknown prefixes should be ignored, they are actually removed. It seems like labels are split on the first space character, and the first part (the prefix) gets removed no matter what. That's why suddenly there are duplicates, which should not happen (`Impedance-0` and `Impedance-1` should be `1 Impedance` and `2 Impedance`).

I will take a look and try to fix this.

0220220313, 2021-11-30:
Thanks,hope this problem can be resolved smoothly.

cbrnr, 2021-11-30:
@0220220313 FYI a fix is ready in #10058 (also ping @adam2392 if you are interested).

PR #10058 (2021-11-30): Fix EDF channel type detection
Fixes #10053. The current implementation was broken, because it did not check if a prefix (anything before a space) was valid. This is problematic for channel names containing spaces.

In addition, I've added a new `infer_types` parameter, which by default is `False`. I think this is the better choice, because in general EDF does not impose any structure on channel names.

To do:
- [x] Add test
- [x] Add changelog entry
- [x] Decide when to roll out this change (version 1.0 or in the next backport?)
closed 2021-11-30 by larsoner