mne-tools/mne-python #10057: BUG?: HTML files are big

larsoner, 2021-11-30:
Waiting for CircleCI to upload artifacts, I noticed:
```
Uploading /home/circleci/project/doc/_build/html/generated/mne.chpi.head_pos_to_trans_rot_t.html (107 kB): 
```
locally, it looks like ~80 kB is in a `<ul>` with ~3k lines of the form:
```
   <li class="toctree-l2">
    <a class="reference internal" href="mne.io.anonymize_info.html">
     mne.io.anonymize_info
    </a>
   </li>
```
Looks like this is really the full left column of collapsible API pages.

@drammock have you thought about how this could be optimized at the pydata-sphinx-theme end? We have ~500 of these docs so we are talking about something like 50MB of redundancy in content around these pages, all of which is identical except for which are collapsed/shown (which is maybe easy to take care of in JavaScript?). Maybe it's not worth worrying about, in which case feel free to close.

drammock, 2021-11-30:
This is a known issue upstream. I haven't followed the discussion too closely but I'll look at it again and see what, if anything, can be done about it

larsoner, 2021-11-30:
No rush, good for a Sunday hack! Closing as an upstream issue.

closed 2021-11-30 by larsoner