mne-tools/mne-python #10060: inconsistent behavior: annotate_flat, annotate_muscle_zscore

sappelhoff, 2021-11-30:
`annotate_flat` returns annotations with a proper `orig_time` inherited from `raw`, whereas `annotate_muscle_zscore` returns annotations w ith `orig_time = None`.

In my pipeline I run both functions and then want to "join" both annotation objects like so `annots_bad = annots_flat + annots_muscle` to then later set `raw.set_annotations(annots_bad)`. However, with the current setup, this makes me run into an error:

> ValueError: orig_time should be the same to add/concatenate 2 annotations (got None != 1912-06-22 14:01:10.497486+00:00)

Would the correct fix be to set `annots_flat.orig_time = None` to then enable concatenation? And should we do that in MNE-Python directly, rather than have users do it themselves?

https://github.com/mne-tools/mne-python/blob/6e60792e4f0bffdd66a9fc1fb85c0c45ed802d78/mne/preprocessing/flat.py#L103-L105

https://github.com/mne-tools/mne-python/blob/6e60792e4f0bffdd66a9fc1fb85c0c45ed802d78/mne/preprocessing/artifact_detection.py#L118-L120

hoechenberger, 2021-12-01:
I think they should both contain `orig_time = None`. When attaching this to a `Raw`, the attribute will be set correctly if I'm not mistaken. So I don't see any use in setting it in the returned `Annotations`

sappelhoff, 2021-12-01:
> I think they should both contain orig_time = None. When attaching this to a Raw, the attribute will be set correctly if I'm not mistaken. So I don't see any use in setting it in the returned Annotations

I agree, same issue for annotate_breaks: https://github.com/mne-tools/mne-python/blob/6e60792e4f0bffdd66a9fc1fb85c0c45ed802d78/mne/preprocessing/artifact_detection.py#L525-L531

(annotate_movement has orig_time=None, similar as annotate_muscle_zscore)

hoechenberger, 2021-12-01:
> I agree, same issue for annotate_breaks (which I just used for real data and it's awesome, btw 😉):

Yay! 😁

PR #10067 (2021-12-08): [MRG] FIX: make all annotate_* funcs return annots with consistent orig_time
closes #10060

Now all `annotate_*` functions will return annotations with a consisten `orig_time` (first thought `None` would be a good idea, ... now changing it to `raw.info["meas_date"]` for all)

~~might be worth including in 0.24.1, cc @larsoner~~
closed 2021-12-08 by sappelhoff