mne-tools/mne-python #10070: BUG?: SURE fails to find any dipoles

larsoner, 2021-12-02:
Take mixed_norm_inverse.py and change `meg-eeg-fwd.fif` to `meg-fwd.fif` so that only MEG channels are used, and SURE does not find any dipoles because it selects an alpha that is too large (100; meg-eeg-fwd.fif selects 84). Not sure if this is by design, but it seems like a bug...?

Same problem occurs if you leave it as `meg-eeg-fwd.fif` but `evoked.decimate(2)`, which should be fine given that the data are sampled at 600 Hz and low-passed at 40 Hz already.

agramfort, 2021-12-02:
@PABannier or @QB3 do you have time to look?

QB3, 2021-12-02:
Arg, there are 2 possibilities:
- a bug in the estimation of the SURE
- the SURE requieres the knowledge of the true noise level (that is set to 1 by default because data is assumed to be whitened)

agramfort, 2021-12-03:
this diff works for me @larsoner

```
diff --git a/examples/inverse/mixed_norm_inverse.py b/examples/inverse/mixed_norm_inverse.py
index 5dfca4d791..23eac96783 100644
--- a/examples/inverse/mixed_norm_inverse.py
+++ b/examples/inverse/mixed_norm_inverse.py
@@ -29,7 +29,7 @@ from mne.viz import (plot_sparse_source_estimates,
 print(__doc__)

 data_path = sample.data_path()
-fwd_fname = data_path + '/MEG/sample/sample_audvis-meg-eeg-oct-6-fwd.fif'
+fwd_fname = data_path + '/MEG/sample/sample_audvis-meg-oct-6-fwd.fif'
 ave_fname = data_path + '/MEG/sample/sample_audvis-ave.fif'
 cov_fname = data_path + '/MEG/sample/sample_audvis-shrunk-cov.fif'
 subjects_dir = data_path + '/subjects'
@@ -39,7 +39,7 @@ cov = mne.read_cov(cov_fname)
 # Handling average file
 condition = 'Left Auditory'
 evoked = mne.read_evokeds(ave_fname, condition=condition, baseline=(None, 0))
-evoked.crop(tmin=0, tmax=0.3)
+evoked.crop(tmin=0.05, tmax=0.150)
 # Handling forward solution
 forward = mne.read_forward_solution(fwd_fname)
```

@larsoner this should speed things a bit but honeslty on my machine this example on main runs in 15s for me. With the diff it's 13s. A lot of time is spent on the rest: plotting of dipoles / stc.plot / sparse morph etc.

agramfort, 2021-12-03:
without the plot code this example is just 6s.

larsoner, 2021-12-03:
> this diff works for me @larsoner

Sure, it's not urgent because I found another way to speed things up (below), but I thought it would be worth looking into why it fails to find any dipoles in most of the other cases I tried. I hope others are at least able to replicate the two problematic cases I noted above...?

> @larsoner this should speed things a bit but honeslty on my machine this example on main runs in 15s for me.

Before #10071, which is now in `main`, CircleCI had:
```
    - ../examples/inverse/mixed_norm_inverse.py:                              162.41 sec
```
I was able to drop it to `28.44 sec` on CircleCI by just avoiding the small alphas with `sure_alpha_grid=np.linspace(100, 40, 10)`. So I assume since you said you tested this example on `main` you are benefitting from my changce already :) Remove that and I suspect it'll be (a lot) slower on your system, too, like it was for me (and CircleCI) before #10071.

agramfort, 2021-12-03:
then I missed the commit 🤦 

so we are good here?

larsoner, 2021-12-03:
To me the situation is that SURE fails more often than it works, at least with this example. If that's not considered a bug, then yes we can close.

agramfort, 2021-12-03:
I don't think there is anything to fix here. MxNE is for short time windows and as soon as I reduced the evoked length focussing on the evoked peak it worked fine.

closed 2021-12-03 by agramfort