mne-tools/mne-python #10079: CSP.fit() required y to work, however the parameter is optional in the used mix-in class

hoechenberger, 2021-12-03:
Reported on the forum:

https://mne.discourse.group/t/can-not-do-csp-fit-transform-without-label/4072

Calling `CSP.fit_transform()` without `y` (which is alright according to the API, which defaults `y=None`) yields the following exception:

```python
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/long/anaconda/envs/3.6/lib/python3.6/site-packages/mne/decoding/mixin.py", line 30, in fit_transform
    return self.fit(X, **fit_params).transform(X)
TypeError: fit() missing 1 required positional argument: 'y'
```

This is because the `fit_transform()` method inherited from `TransformerMixin` is not compatible with the `fit` method implemented in `CSP`.

Solution is to give `CSP` its very own `fit_transform` method.

larsoner, 2021-12-03:
I think our typical solution is just to put y=None in the fit method, then never use it

hoechenberger, 2021-12-03:
Since CSP is supervised, it cannot work without `y`.

PR #10084 (2021-12-03): MRG: Make y a required parameter in CSP.fit_transform()
Fixes #10079
commit c1cbc784 (2021-12-03): MRG: Make y a required parameter in CSP.fit_transform() (#10084)

* Make y a required parameter in CSP.fit_transform()

Fixes #10079

* Update mne/decoding/csp.py [skip azp] [skip actions]
closed 2021-12-03 by larsoner