mne-tools/mne-python #10083: Vertical placement of channel traces seems off in raw butterfly plot

hoechenberger, 2021-12-03:
MWE:
```python
# %%
from pathlib import Path
import mne
import numpy as np

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(sample_fname, preload=True)

# %%

fig = (
    raw.copy()
       .pick_types(eeg=True, eog=True, stim=True)
       .plot(butterfly=True, show_scrollbars=False, start=0,
             duration=1, show=False)
)
```
[image]

It appears as if the stim channel has a lot of signals, and the EOG channel were flat.

However, the following code reveals that this ain't so:
```python
# %%
fig = (
    raw.copy()
       .pick_types(stim=True)
       .plot(butterfly=True, show_scrollbars=False, start=0,
             duration=1, show=False)
)
```
[image]

```python
# %%
fig = (
    raw.copy()
       .pick_types(eog=True)
       .plot(butterfly=True, show_scrollbars=False, start=0,
             duration=1, show=False)
)
```
[image]

cc @drammock

larsoner, 2021-12-03:
> It appears as if the stim channel has a lot of signals, and the EOG channel were flat.

How do you know which is supposed to be which? To me I see three sets of signals, which I'd expect. But I wouldn't expect it to be in the order you gave in `pick_types` because that function does not reorder channels, or set the plotting order.

If you want specifically ordered channels, you need to pass `order=...`, probably in your case something like:
```
order = np.concatenate([
    mne.pick_types(raw.info, eeg=True),
    mne.pick_types(raw.info, eog=True),
    mne.pick_types(raw.info, stim=True)])
raw.plot(order=order)
```
gets you closer to what you want. However, I'm not sure if butterfly will respect this, it might just be non-butterfly mode.

If this is the case, I would just update the doc for the `order` param to say that it isn't used in butterfly mode, because butterfly mode has to group by channel type. (I guess we could go based on the order we see each channel type *first in the instance*, but I think this has disadvantages, too, so probably shouldn't be the default.)

hoechenberger, 2021-12-03:
> How do you know which is supposed to be which?

Based on the ordinate labels. The respective butterfly traces should oscillate around the respective channel type label, or be cropped if they're exceeding a certain range.

It's misleading if, like in the first picture, the `eog` label is perfectly aligned with a flat line, when in fact this line represents the `stim` channels.

hoechenberger, 2021-12-03:
@larsoner Will you be around at today's dev meeting? We can discuss there.

larsoner, 2021-12-03:
Oh wow I didn't even see the labels because of image transparency and GitHub dark mode. Agreed those labels are wrong :)

larsoner, 2021-12-03:
No I can't, but it seems like a straightforward bug to me. The fix is hopefully similarly straightforward, but with viz, who knows!

hoechenberger, 2021-12-03:
> No I can't,

Bummer!

> The fix is hopefully similarly straightforward, but with viz, who knows!

Yeah I have no idea honestly :) Maybe @drammock has an immediate intuition?

PR #10087 (2021-12-03): fix trace offsets in butterfly mode
closes #10083
closed 2021-12-03 by drammock