mne-tools/mne-python #10088: Fast annotation from mask

snwnde, 2021-12-03:
#### Describe the new feature or enhancement
In `mne/preprocessing/artifact_detection.py` there is a function called `_annotations_from_mask`. As its name suggests it creates an annotation from a numpy array mask. A for loop has been used in its implementation, which could be very time-consuming when the number of components is large.

#### Describe your proposed implementation
This could be implemented without any for loop. I have the following code, working in my tests:
```python
def _annotations_from_mask(times, mask, annot_name):
    """Construct annotations from boolean mask of the data."""
    from scipy.ndimage.morphology import distance_transform_edt
    from scipy.signal import find_peaks
    mask_tf = distance_transform_edt(mask)
    mid_width_pt = find_peaks(mask_tf)[0]
    onsets = times[mid_width_pt - mask_tf[mid_width_pt].astype(int)]
    ends = times[mid_width_pt + mask_tf[mid_width_pt].astype(int)]
    durations = ends - onsets
    desc = [annot_name]*len(durations)
    return Annotations(onsets, durations, desc)
```
This implementation doesn't require any additional package.
With a mask of `num_comps` equal to 24389, `%timeit` gives for the original implementation and the mine respectively:
```python
2min 1s ± 640 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```
```python
590 ms ± 16.2 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

#### Describe possible alternatives
I do not have any other implementation in mind.

Please tell me how you think of this enhancement. If everything seems fine, I will be glad to create a PR.

agramfort, 2021-12-04:
looks smart !

can you send us a PR?

thx a lot

>

snwnde, 2021-12-04:
I tweaked the code to address the problem that `find_peaks` cannot find peaks at the start and the end of arrays.

```python
def _annotations_from_mask(times, mask, annot_name):
    """Construct annotations from boolean mask of the data."""
    from scipy.ndimage.morphology import distance_transform_edt
    from scipy.signal import find_peaks
    mask_tf = distance_transform_edt(mask)
    # Overcome the shortcoming of find_peaks
    # in finding a marginal peak, by
    # inserting 0s at the front and the
    # rear, then subtracting in index
    ins_mask_tf = np.concatenate((np.zeros(1), mask_tf, np.zeros(1)))
    midpoint_index = find_peaks(ins_mask_tf)[0] - 1
    onsets_index = midpoint_index - mask_tf[midpoint_index].astype(int) + 1
    ends_index = midpoint_index + mask_tf[midpoint_index].astype(int) - 1
    # Ensure onsets_index >= 0,
    # otherwise the duration starts from the beginning
    onsets_index[onsets_index < 0] = 0
    # Ensure ends_index < len(times),
    # otherwise the duration is to the end of times
    ends_index[ends_index >= len(times)] = len(times) - 1
    onsets = times[onsets_index]
    ends = times[ends_index]
    durations = ends - onsets
    desc = [annot_name] * len(durations)
    return Annotations(onsets, durations, desc)
```

agramfort, 2021-12-04:
Can you send the pull request or shall we do it ?

snwnde, 2021-12-04:
> Can you send the pull request or shall we do it ?

I created the PR #10089. Some checks fail though. It seems a precision problem.

closed 2021-12-06 by snwnde