mne-tools/mne-python #10090: Add probabilty in get_montage_volume_labels() with one roi

BarryLiu97, 2021-12-05:
Hi, the ```get_montage_volume_labels``` now give us all the regions in like '''aparc+aseg'''. In my opinion, calculate the probabilty, for example, in bipolar SEEG, we could see a electrode is a aphere with 5 mm radius, about 175 voxels, then calculating how many voxels are ina region, the choose the region with the biggest probability as teh anatomy of this electrode.

agramfort, 2021-12-05:
@alexrockhill ?

alexrockhill, 2021-12-06:
@BarryLiu97 5mm is a pretty large electrode contact, there are ideas like https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0242704 to model the deformation of the tissue surrounding the electrode with biomechanics. I think for sEEG these effects are small but on the CT where the electrode is actually displaces the brain tissue and isn't really the location of the tissue being recorded.

I think you would want to do something with a finite element model using the bioelectric properties of gray and white matter to estimate where the source of the recording is coming from but that's quite complicated and may even be beyond the scope of MNE-Python (i.e. would probably be better in a new mne-tools package called something like MNE-FEM or MNE-iEEG). I think it's a good idea, but I'm not sure how you assign a probability to all the labels of contacts other than just their distance. Technically this is possible now as you could use `get_montage_volume_labels` at different distances and assign the label based on these somehow. This could be improved by a PR to facilitate not only a label for each contact but also a distance possibly.

BarryLiu97, 2021-12-06:
Hi, I should make my opinion more clear. Actually, this probability is not a real "probability", it's more like, given a distance, how many voxels are in each neighbour region. It;s more like a rate. I got this idea through reading the tutorial in brainstorm [tutorial](https://neuroimage.usc.edu/brainstorm/Tutorials/Epileptogenicity#Anatomical_labelling). section Anatomical labelling.
In fact, when I set dist =1 or 2, the result are much accurate than 5.

alexrockhill, 2021-12-06:
Ah okay, I gave it a brief read and to my knowledge both tools are performing the same computations. 5 mm in any direction is a very large distance relatively.

Another option is to use the `elec_image` from `warp_montage_volume` which represents the positions of the contacts in voxels (where ever the image is brighter than some threshold) optionally with a binary dilation to get the next-nearest voxels are well (see [`scipy.ndimage.](https://docs.scipy.org/doc/scipy-0.14.0/reference/generated/scipy.ndimage.morphology.binary_dilation.html)). I will actually make an issue to expose that more easily as I think the `elec_image` is helpful to work with. Note this is not what brainstorm is doing, what is already implemented matches what brainstorm is doing, this would be an alternate voxel-based approach that may be helpful especially for contacts of different sizes.

BarryLiu97, 2021-12-06:
Thank you for your patience!

closed 2021-12-10 by alexrockhill