mne-tools/mne-python #10099: VTK version with python 3.9 and windows

mscheltienne, 2021-12-06:
I have set up a new venv on a different windows laptop for MNE and I wanted to move from Python 3.8.10 to 3.9.9. 
When doing `py -m pip install -r requirements.txt`, the installation fails because there are no wheels on Pypi for `vtk<=9.0.1` for windows/Python 3.9.
I installed version 9.1.0 instead; and for now, at least, I didn't notice any issues with MNE. I couldn't run yet the test suite as my battery died.. but if you have some tests in mind to make sure version 9.1.0 works fine; I can run them tomorrow. 

Maybe the requirement `vtk<=9.0.1; platform_system == "Windows"` is outdated?

larsoner, 2021-12-06:
+1 for a quick PR to make this change and see if CIs come back happy. It's probably there because CIs are problematic. Did you look at `git blame`?

https://github.com/mne-tools/mne-python/blame/main/requirements.txt

Looks like it came from a PR to fix Mayavi: https://github.com/mne-tools/mne-python/pull/9512

So I'm at least a bit optimistic that we can remove this limitation now that we don't use mayavi...

mscheltienne, 2021-12-06:
Let's see if something breaks, I was on the go and didn't have time to check more. 
I didn't think about using the online CIs with a PR, that was too simple 😅

PR #10100 (2021-12-08): Remove constrain on vtk version for windows
Test if VTK version still needs to be constrained for Windows.

Closes #10099
closed 2021-12-08 by larsoner