mne-tools/mne-python #10101: [ENH] Expose `elec_image` from `mne.warp_montage_volume`

alexrockhill, 2021-12-06:
Related to https://github.com/mne-tools/mne-python/issues/10090.

In working with my own data, we have contacts that are different sizes and using a strict distance-based metric of labeling as implemented in `mne.get_montage_volume_labels` sometimes returns unhelpful labels at relatively low distances (e.g. 3 mm crossing the dura and being labeled as cortex in the cerebellum or having labels in the brainstem from subcortical structures). The function to compute which voxels are within a threshold of the maximum is already implemented in `mne.get_montage_volume_labels` and actually the input to that function is very cumbersome. Breaking it down into two steps of making an image in voxels and then warping that image I think would be easier to understand from a user perspective. Since there has been a release since this function was added the deprecation will be a bit complicated though. Also an option is not to deprecate and just call the montage-to-image function from `warp_montage_volume` but I think it would be good to split that up for the long-term maintainability but just my opinion.

larsoner, 2021-12-07:
> Also an option is not to deprecate and just call the montage-to-image function from warp_montage_volume but I think it would be good to split that up for the long-term maintainability but just my opinion.

I would say let's start with the split-up-but-don't-deprecate plan. Maybe `get_montage_volume_labels` can be saved someday if people find a good reliable algorithm that works 90%+ of the time. And if not, it can at least serve as a quick pass version.

alexrockhill, 2022-01-07:
Where I was going with this is that BIDS has electrode contact size as a parameter but I think it might be better instead of exposing the image to make a function to calculate it. Here is where it is described in the spec: https://bids-specification.readthedocs.io/en/stable/04-modality-specific-files/04-intracranial-electroencephalography.html. I think ideally, you would know the size but it would be better to have an estimate than nothing at all. This might be an mne-bids issue but I think this is really tricky because there isn't anywhere in the `Raw` object to store the size as far as I know.

alexrockhill, 2022-01-07:
> Maybe `get_montage_volume_labels` can be saved someday if people find a good reliable algorithm that works 90%+ of the time. And if not, it can at least serve as a quick pass version.

I was thinking about this and there are biomechanical models of brain tissue deformation with a needle insertion into a brain 10.1371/journal.pone.0242704 but this is quite advanced and doesn't seem like the biggest deal for sEEG. The ground truth is that the electrode contact displaces the tissue so I think it wouldn't make sense for larger contacts to have more labels based on volume in the CT, surface area is what is important. I guess unless you have model of the voxel displacement, the best approach available is to just pick voxels within a certain distance of the center. This could be adjusted based on the size of the contact by the user by picking specific channels to compute labels for at a time. This does make assigning the channels sizes, preferably in the `Raw` object something that would be helpful for ieeg users.

alexrockhill, 2022-01-31:
I think this is incorporated into #10216 and has to do with actually modeling what's going on with electrical activity in the vicinity of the contact and not just treating it as a point.

closed 2022-01-31 by alexrockhill