mne-tools/mne-python #10106: `mne.Epochs` sends wrong log messages (1) on metdata, (2) when `decim` is passed

sappelhoff, 2021-12-07:
`mne.Epochs` sends wrong log messages (1) on metdata, (2) when `decim` is passed.

MWE:

Try first running this code:

```Python
import os
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=120)
raw.load_data()
raw.filter(h_freq=40, l_freq=None)

events = mne.find_events(raw, stim_channel='STI 014')
event_dict = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
              'visual/right': 4, 'face': 5, 'button': 32}
epochs = mne.Epochs(raw, events, tmin=-0.2, tmax=0.5, event_id=event_dict,
                    decim=1, preload=True)
print(epochs.times.shape)

```

This is the logged output, where I can see that the `Not setting metadata` log is unnecessarily repeated.
On the last lines of the log, we see that 421 time points were loaded, and this is indeed the shape of `.times`

```text
176 events found
Event IDs: [ 1  2  3  4  5 32]
Not setting metadata
Not setting metadata
176 matching events found
Setting baseline interval to [-0.19979521315838786, 0.0] sec
Applying baseline correction (mode: mean)
Created an SSP operator (subspace dimension = 3)
3 projection items activated
Loading data for 176 events and 421 original time points ...
1 bad epochs dropped
(421,)
```

Now change `decim=1` in the `mne.Epochs()` call above to `decim=2`. I would expect the log 

> Loading data for 176 events and 421 original time points ...

to change in terms of "original time points" loaded (should be half as many)

However, this is what we get:

This is the logged output, where the shape of `.times` has indeed halved as expected, yet the log tells us that 421 time points were loaded.

```text
176 events found
Event IDs: [ 1  2  3  4  5 32]
Not setting metadata
Not setting metadata
176 matching events found
Setting baseline interval to [-0.19979521315838786, 0.0] sec
Applying baseline correction (mode: mean)
Created an SSP operator (subspace dimension = 3)
3 projection items activated
Loading data for 176 events and 421 original time points ...
1 bad epochs dropped
(211,)

```

# Needed actions

1. remove duplicate log message on setting metadata. Don't really know where to fix, maybe around here: https://github.com/mne-tools/mne-python/blob/a3066f6eb3730d13d2dce632ff6cb5d4870fbe7f/mne/utils/mixin.py#L412-L418

2. adjust logging about loaded time points when creating epochs and decimating. Probably to be fixed somewhere around here: https://github.com/mne-tools/mne-python/blob/a3066f6eb3730d13d2dce632ff6cb5d4870fbe7f/mne/epochs.py#L1388-L1389

larsoner, 2021-12-08:
> to change in terms of "original time points" loaded (should be half as many)

IIRC "original" here I think is meant to mean "pre-decimated" / "raw". Maybe we should just say "raw" since it does load that many time points and then they are decimated...?

sappelhoff, 2021-12-08:
I see :thinking: that makes sense. Does Epochs also "load" these data if the `raw` I pass is already pre-loaded?

> Maybe we should just say "raw" since it does load that many time points and then they are decimated...?

or - if `decimate>1` - modify the log to say:

> Loading data for 176 events and 421 original time points **(prior to decimation)** ...

larsoner, 2021-12-08:
> Does Epochs also "load" these data if the raw I pass is already pre-loaded?

Not sure actually

> or - if decimate>1 - modify the log to say:

+1 for me, would make things less clearer

larsoner, 2021-12-08:
> less unclear?

Yes sorry, I originally wrote "less ambiguous" and went to replace it with "clearer" but only deleted part of the original :)

> I believe the first call can be deleted, WDYT?

I think probably yes? I would try making the change and see if any test breaks. Sometimes in our code we do have more than one assignment like this (where it seems like one is redundant) but it's necessary for some weird reason like private function calling order... but usually in these cases we try to leave a comment saying "this looks redundant but it's necessary because ...". If there is no such comment, it's possibly just cruft / errant double-setting.

It would be good for example for TDD to make it so that you
```
with catch_logging() as log:
    Epochs(..., verbose=True)
log = log.getvalue()
assert log.count('Setting metadata') == 1
```
or so

sappelhoff, 2021-12-08:
> I think probably yes? I would try making the change and see if any test breaks. Sometimes in our code we do have more than one assignment like this (where it seems like one is redundant) but it's necessary for some weird reason like private function calling order.

it seems like that's the case :thinking: I already pushed the commits to a PR (bad practice, sorry). but testing this locally reveals that with the removal of the "apparent" unneeded duplicate of `self.metadata = metadata`, at least one test fails.

If this were setting a value via some function, I would propose to add a `do_log=False` to that function call, so that only the second log message gets sent. But for this case, I don't know right now.

larsoner, 2021-12-08:
> If this were setting a value via some function, I would propose to add a do_log=False to that function call, so that only the second log message gets sent. But for this case, I don't know right now.

In principle you can do something like
```
if orig_metadata is not None:  # or whatever this conditional needs to be
    ctx = partial(use_log_level, verbose='warn')
else:
    ctx = nullcontext()
with ctx:
    self.metadata = metadata
```
But we can do it more simply than this. Note that the `metadata.setter` already has a `verbose` kwarg:

https://github.com/mne-tools/mne-python/blob/b7daa3603d75062f713c84707da3a940c7de7266/mne/utils/mixin.py#L400-L402

So we can use the fancy fset call like here:

https://github.com/mne-tools/mne-python/blob/b7daa3603d75062f713c84707da3a940c7de7266/mne/utils/mixin.py#L218):

sappelhoff, 2021-12-09:
Thanks @larsoner for these suggestions. That will fix part of the problem: When no metadata is set, people will get the correct log.

However, when metadata **is** being set, people should see one log:

`Adding metadata with X columns`

Instead, they will see:

`Replacing existing metadata with X columns`

because internally, metadata will already be set from the first time (the one time we didn't log about).

sappelhoff, 2021-12-09:
Okay I think nothing is stopping me from just deleting the existing metadata and then setting it again :thinking:  I'll just try it

larsoner, 2021-12-09:
Or add a msg kwarg to the setter so you can say Adding instead of Replacing

PR #10112 (2021-12-13): [MRG] FIX: Clearer (unambiguous) logging during Epochs instantiation
closes #10106

~~Should I add a whatsnew entry? Or is this too minor?~~
closed 2021-12-13 by larsoner