mne-tools/mne-python #10107: Looking for help to improve Report.add_raw() / raw.plot() performance

hoechenberger, 2021-12-07:
I've been trying to improve speed of Report generation. One of the things that IMHO takes too long generation of the butterfly plots in `Report.add_raw()` (if `butterfly=True`, which is the default; this will generate 10 butterfly plots of 1-second segments of the data). With the help of @agramfort I profiled this and visualized the results with [SnakeViz](https://jiffyclub.github.io/snakeviz/).

I ran the following code in an IPython window:

```python
from pathlib import Path
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname, verbose=False)
report = mne.Report()

# %%
%prun -D report.prof report.add_raw(raw=raw, title='foo', psd=False, replace=True)
```

This generates the following output:

[image]

As you can see in the 6th and 7th row (counting from the top), half of the time of `raw.plot()` isn't actually spent on **plotting**, but rather on setting up the plotting window.

I could not really figure out why this is taking so long, it seems lots of time is spent in Matplotlib's `add_path()`, which to my knowledge is used to draw the scrollbars; however, `Report.add_raw()` passes `show_scrollbars=False`, so I'm not sure if this is related to something else here …

If anyone's got an idea how to improve things (@larsoner, @drammock?), please do let me know!

larsoner, 2021-12-07:
I would try doing this generation after picking one channel of each type (MEG, EEG, stim, ECG) from `sample`. If it's super fast, and the full-ch-count is slow, we know it's the actual trace drawing that is painful. If it's still slow, we can dig deeper into why setting up the draw takes so long.

hoechenberger, 2021-12-07:
@larsoner 
This is what I get when running the same code as above, but with an additional `raw.pick_types(eeg=True)`, i.e., only plotting EEG channels:

[image]

Figure setup still takes a considerable amount of time.

hoechenberger, 2021-12-07:
I also tried with preloading the data; while this reduces total run time (apparently mostly because `_compute_scalings()` takes much less time then), `_get_browser()` run time is unchanged.

larsoner, 2021-12-08:
> > I would try doing this generation after picking one channel of each type (MEG, EEG, stim, ECG) from sample

>  but with an additional raw.pick_types(eeg=True)

This isn't quite what I meant -- I was suggesting *one channel of each type* (e.g., `pick_channels(['MEG 0111', 'MEG 0112', 'EEG 001', 'ECG 001', 'STI 014'])` or so) rather than *all channels of one type*. What I suggested increases the greatest contrast between number of traces (minimized) versus number of things that need to be set up (ticks, positions, etc.; maximized/maintained). By picking all EEG channels you've reduced the number of traces (~400 -> ~60) but you've also reduced the number of things that need to be set up (~5 channel types to 1), so doesn't isolate the "setup matplotlib figure and axes and ticks etc." step quite as well.

hoechenberger, 2021-12-08:
Oh, sorry for the misunderstanding.

Here is what it looks like with just a single channel – spoiler: pattern is the exact same as above!
```python
# %%
from pathlib import Path
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname, verbose=False)
raw.pick([0])
report = mne.Report()

# %%
%prun -D report.prof report.add_raw(raw=raw, title='foo', psd=False, replace=True)
```
[image]

hoechenberger, 2021-12-08:
So here is what it looks like with a single channel for each channel type. Spoiler: pattern remains the same:

```python
# %%
from pathlib import Path
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname, verbose=False)
# raw.pick([0])
raw.pick_channels(['MEG 0111', 'MEG 0112', 'EEG 001', 'ECG 001', 'STI 014'])
report = mne.Report()

# %%
%prun -D report.prof report.add_raw(raw=raw, title='foo', psd=False, replace=True)
```
[image]

And here with just a single channel (i.e., following `raw.pick([0])`):
[image]

larsoner, 2021-12-08:
One easy pain point to fix is `_compute_scalings` (at least for non-preloaded data), which shouldn't need to be called by default (it will read all data, which is expensive). We shouldn't use `scalings='auto'`, I'd rather expose a `scalings` argument to `add_raw` that defaults to `None` like for `raw.plot`.

hoechenberger, 2021-12-08:
Good idea!

Exposing the param shouldn't be necessary, **however** doing that would resolve the following feature request – 2 birds, one stone!

https://github.com/mne-tools/mne-python/pull/10105#issuecomment-988254943

hoechenberger, 2021-12-08:
With precomputed scaling, I can reduce runtime of the MWE above (1 chan per channel type) by about 40%

[image]

Will test on a "real" dataset now and make a PR. Thanks for your help so far, @larsoner, you once again had a good intuition! Looking forward to doing the next few iterations ;)

larsoner, 2021-12-08:
It is still not awesome that it takes ~3.5 seconds to draw a figure with only 5 traces in it (1.27 for figure creation and 2.12 to actually draw). I'll try to take a look, it seems like we should be able to cut that down.

> #10105 (comment)

This might actually get worse with the default scalings, we'll have to see. In that case, adding a `scalings` param is an easy solution.

hoechenberger, 2021-12-08:
I'm currently drafting a first PR in #10109

hoechenberger, 2021-12-08:
I managed to further reduce processing time of a real-world dataset (8 runs of MEG data) from 227 to 145 seconds on our institute's server by disabling visualization of annotations in the butterfly plots. Will file a PR for this one shortly.

closed 2021-12-20 by hoechenberger