mne-tools/mne-python #10121: Read Net Stations' ECI events information

fncokg, 2021-12-12:
When using Net Station and E-prime together, events are often recorded and sent via TCP-IP under its ECI(Experiment Control Interface), which are then saved in `foo.mff/Events_ECI TCP-IP 55513.xml`. But when the whole `.mff` file is read via `mne.io.read_raw_egi()`, this information is not imported and I haven't seen any efficient method to do this.

agramfort, 2021-12-12:
either we fix the reader or you just write some code to parse this file and
get events or annotations.

is there available online a file with this so we can use it for testing?

if you write such parsing code can you share it with us?

Alex

fncokg, 2021-12-13:
@agramfort Thanks for your reply.

> is there available online a file with this so we can use it for testing?

Yes, online testing files have existed in the `mne-testing-data` repo ,such as in (this file)[https://github.com/mne-tools/mne-testing-data/blob/master/EGI/test_egi_multiepoch_eprime.mff/Events_ECI%20TCP-IP%2055513.xml].

> if you write such parsing code can you share it with us?

I've already written a version for my use case and I'd like to submit a PR to you. Only built-in `xml` package is used so that there are no more dependencies. Where do you think is good to put this function? Two possible considerations:

- In `RawEGI.__init__()`, search if there's a XML file like `Events_ECT*.xml`. If there is, read it, convert it to an `Annotations` obj and save it in a special attribution such as `RawEGI.ect_events_annotations`.
- Define a new method in `RawEGI` such as `RawEGI.read_ect_events()` and return the `Annotations` obj directly.

Which one do you think is good? Or any other new idea?

agramfort, 2021-12-13:
I would read these as store them as extra annotations in raw.annotations
via raw.set_annotations

>

fncokg, 2021-12-13:
> I would read these as store them as extra annotations in raw.annotations via raw.set_annotations
> […](#)

That's a good idea.

obergmartin, 2023-11-03:
Sorry for reviving an old thread, but I've been looking though your fork, @fncokg , and can't find how you solved this. I'm using MNE1.4.2 and want to import the data in the Events_ECI*.xml file.

fncokg, 2023-11-04:
> Sorry for reviving an old thread, but I've been looking though your fork, @fncokg , and can't find how you solved this. I'm using MNE1.4.2 and want to import the data in the Events_ECI*.xml file.

Sorry for that. I might forget to push the codes. Now I've pushed them to my fork. You can check the comments of the `read_raw_egi` function in the `mne\io\egi\egi.py` file to get detailed information of my implementation. I'm recently busy on my phd thesis and have no time to do the necessary tests, and therefore I didn't submit a PR to the main repo for 2 years :)

closed 2023-11-14 by fncokg