mne-tools/mne-python #10128: Scaling label overplotting in Raw.plot()

hoechenberger, 2021-12-14:
MWE:
```python
# %%
from pathlib import Path
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname)
raw.plot(butterfly=True, start=0, duration=1)
```
Take a look at the scaling label ob the `grad` channels:
[image]

Interestingly, when only plotting MEG channels, this problem does not appear:
```
# %%
raw.pick_types(meg=True)
raw.plot(butterfly=True, start=0, duration=1)
```
[image]

cc @drammock

drammock, 2021-12-14:
the "background noise" is a bad channel, and is missing in the second example because `pick_types` defaults to `exclude='bads'`.  The overplotted scale bars are indeed a problem though, will look into it.

hoechenberger, 2021-12-15:
@drammock LMK if you could need a hand here!

drammock, 2021-12-15:
> @drammock LMK if you could need a hand here!

if you have bandwidth to look at this, I don't currently; I might on friday?

hoechenberger, 2022-01-04:
Sorry, I didn't have time to look into this either :(

PR #12721 (2024-07-17): BUG: Fix bug with scale bar overplotting
Closes #10128

Test fails on `main` and passes on this PR. And locally testing visually I now get:
```
MNE_BROWSER_BACKEND=matplotlib mne browse_raw ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif --butterfly
```

[image]

Also adds a `--butterfly` option to `mne browse_raw` but I think this is probably not worth adding to the changelog because it's so minor.
closed 2024-07-17 by drammock