mne-tools/mne-python #10147: issue importing fNIRS data in mne

mlrocca, 2021-12-17:
#### Describe the new feature or enhancement
*Please provide a clear and concise description of what you want to add or
change.*

I am trying to import my fNIRS data with
mne.io.read_raw_nirx("/Users/name/Downloads/PROTOCOLLO_TDCS_FILE_GREZZI/SHAM/CONTROLLI/2018-11-27_002/")
but I have this error message

Loading /Users/name/Downloads/PROTOCOLLO_TDCS_FILE_GREZZI/SHAM/CONTROLLI/2018-11-27_002/
<ipython-input-8-14238638d873>:1: RuntimeWarning: Only import of data from NIRScout devices have been thoroughly tested. You are using a "NIRSport 8x8" device. 
  mne.io.read_raw_nirx("/Users/name/Downloads/PROTOCOLLO_TDCS_FILE_GREZZI/SHAM/CONTROLLI/2018-11-27_002/")
<ipython-input-8-14238638d873>:1: RuntimeWarning: Extraction of measurement date from NIRX file failed. This can be caused by files saved in certain locales. Please report this as a github issue. The date is being set to January 1st, 2000, instead of "mar 27 nov 2018""18:13:22.559"
  mne.io.read_raw_nirx("/Users/name/Downloads/PROTOCOLLO_TDCS_FILE_GREZZI/SHAM/CONTROLLI/2018-11-27_002/")
Out[8]: <RawNIRX | 2018-11-27_002, 40 x 1934 (247.4 s), ~54 kB, data not loaded>

We used a continuous wave NIRS system (NIRSport 8X8, Nirx Medical Technologies LLC, Berlin, Germany). The fNIRS data acquisition software was the NIRStar 14.2 (Version 14, Revision 2, Release Build, 2016-04-15 NIRx Medizintechnik GmbH, Berlin, Germany; www.nirx.net).

#### Describe your proposed implementation
*Describe how you think the feature or improvement should be implemented (e.g., 
as a new method on an existing class? as new capability added to an existing
method?) If you're not sure, please delete this section and the next section.*

 Please, could you help me to implement a function that can import my data?

#### Describe possible alternatives
*If you've suggested an implementation above, list here any alternative
implementations you can think of, and brief comments explaining why the chosen
implementation is better.*

#### Additional comments
*Add any other context or screenshots about the feature request here.*

larsoner, 2021-12-17:
Can you share the problematic file? Or even better, any chance you could record a little 1 sec data file from this system and share that so we could add it to our tests?

larsoner, 2021-12-17:
cc @rob-luke

mlrocca, 2021-12-17:
Sure,

Please find enclosed an example of data that gives me that issue
[2020-01-24_SHAM_CTRL_0050.zip](https://github.com/mne-tools/mne-python/files/7738037/2020-01-24_SHAM_CTRL_0050.zip)

I am looking forward to hearing from you
.

rob-luke, 2021-12-20:
Hi @mlrocca,

Thanks for this very clear report. MNE currently only supports reading NIRx data from NIRStar version 15.0 and above. Whereas, you are trying to read data acquired with NIRStar version 14.2. I don't think the file format changed very much, so adding support might not be too much effort. Could you open a PR to add support for older NIRx files?

rob-luke, 2021-12-20:
This might also be related to https://github.com/mne-tools/mne-python/pull/7891 where we had issues handling dates with formats other than Australian English.

mlrocca, 2021-12-20:
Thank you, 

I opened a new issue to add support for older NIRx versions.

If there is also the issue of the dates, is it possible to fix that too?

Best,

mlrocca, 2021-12-20:
Ok thank you,

how long do you think you will take to implement that feature?

Are you going to advice me when it will be implemented?

drammock, 2021-12-20:
> Ok thank you,
> 
> how long do you think you will take to implement that feature?

I don't know.  The winter holidays mean many of our developers are not available right now.  

> Are you going to advice me when it will be implemented?

The pull request that fixes the issue will be linked here and will automatically close this issue when it is merged in.  Unless you've disabled it in your GitHub notification settings, you will get a notification when that happens.

mlrocca, 2022-01-31:
Hi,

any update on the implementation of the feature in the object?

PR #10277 (2022-02-01): BUG: Fix bug extracting meas_date
Closes #10147

Turns out that localized dates is a pain in Python, and solving it in a general way requires a lot of work and/or additional dependencies. This implements a simpler solution: use a script to make a dict of translations and try them all. It's quick to execute, and easy enough to generate the dict of translations that we need.

@mlrocca feel free to try this branch (or just wait until it's merged into `main` and try that), it should fix your bug.

@rob-luke feel free to review and merge if you're happy
closed 2022-02-01 by agramfort