mne-tools/mne-python #10148: Provide stand-alone MNE-Python installers

hoechenberger, 2021-12-17:
This is currently a WIP carried out at https://github.com/hoechenberger/mne-installers

Would be great to get this tested & ready for MNE-Python 1.0

hoechenberger, 2021-12-17:
Related issue: https://github.com/conda/constructor/issues/471

rob-luke, 2021-12-20:
I just tested the stand alone installer on my Intel Mac and it worked perfectly. I was able to run the fNIRS tutorial including obtaining the 3D plots.

hoechenberger, 2021-12-20:
Windows installers are now also tested on CI, next step is to get a person to run this on their computer ;)

hoechenberger, 2021-12-20:
Will test this with Guillaume on Wednesday

larsoner, 2022-02-17:
@hoechenberger when you get a chance, could you open a PR to MNE-Python to show how you would change the installation docs? That way people can look at the updated instructions and try them out (substituting the current prerelease executable names for the final versions) and we can also work on getting the wording right.

hoechenberger, 2022-02-21:
I will work on revised install docs tomorrow.

larsoner, 2022-02-23:
@hoechenberger are we good to go here, other than Dipy + Python 3.10 if we can get it? I say we wait a few more days to see if they do cut a release, but ultimately the existing `dipy` + 3.9 works fine

hoechenberger, 2022-02-23:
Yes, sorry, I'm behind schedule with the updated docs. But the installer seems to work. I do have a list of things I'd like to improve in the future but there are no blockers. I've tested it on Intel Macs and on Linux, and @GuillaumeFavelier has tested it on Windows 10. I'll test on an M1 Mac this Friday (yes, I know, I've promised to do this for weeks now), but I don't expect any issues.

hoechenberger, 2022-02-23:
@larsoner If you want to give it a try: See the assets provided here: https://github.com/hoechenberger/mne-installers/releases/tag/v1.0.0-pre5

larsoner, 2022-03-01:
@hoechenberger it looks like there has been some dipy activity moving them closer to release, so it would be good to work on the updated instructions soon

hoechenberger, 2022-03-01:
on my todo for tomorrow!

hoechenberger, 2022-03-04:
@larsoner Would you be available for a review on Monday? Still WIP at my end but I think Monday I will have something to share :)

larsoner, 2022-03-04:
Yep

larsoner, 2022-03-11:
FYI Dipy 1.5 is out! https://pypi.org/project/dipy/1.5.0/

hoechenberger, 2022-03-12:
😱😱😱

larsoner, 2022-03-15:
Just a reminder that, as part of this, we should update https://github.com/hoechenberger/mne-installers/blob/main/README.md to hopefully just point people to https://mne.tools/dev/install/index.html

PR #10436 (2022-03-19): MRG: Add installer documentation
I was unhappy with what I had so far and started from scratch again (sorry!)

This is the current state of the docs

What's missing:

- [ ] What to do **after** installation?
- [ ] It should be easier to directly navigate to the instructions for the desired platform (Windows, Linux, macOS) from the top of the page? Maybe add buttons? Links to anchors?
- [ ] Include some screenshots – or maybe just one, at the top somewhere. I have plenty from macOS.
- [ ] Re-arrange the "old" install docs
- [ ] Add mne-kit-gui (assuming https://github.com/mne-tools/mne-installe [... cut]
closed 2022-03-19 by agramfort