mne-tools/mne-python #10149: Clarification on the relation between SNR and regularization factor

mh105, 2021-12-18:
#### Proposed documentation enhancement
Could we reconcile the different mentions of the relation between SNR and regularization factor for inverse solution? Following are couple of places that don't agree with each other: 

- https://mne.tools/stable/overview/implementation.html#whitening-and-scaling says \lambda^2 ~ 1 / SNR
- https://github.com/mne-tools/mne-python/blob/c3ff05c05c7bfcf765249f27a517803f27f48b4e/mne/minimum_norm/inverse.py#L532 recommends \lambda^2 ~ 1 / SNR^2 
- https://mne.tools/stable/auto_examples/inverse/mixed_source_space_inverse.html?highlight=snr#compute-inverse-solution also uses \lambda^2 ~ 1 / SNR^2
- https://mne.tools/stable/generated/mne.minimum_norm.estimate_snr.html?highlight=snr#mne-minimum-norm-estimate-snr docstring says \lambda ~ 1 / SNR^2

It will be helpful too if there's a reference for this relation.

agramfort, 2021-12-19:
good poing @mh105 

I think the confusion comes from the implicit definition of SNR in each context. Is the SNR in power (ie amplitude squared) or in amplitude.

this "lambda2 = 1/snr**2" comes from the historical convention in MNE-C code I think.

@mshamalainen any thought?

@larsoner can you find some historicals refs to this in the C code?

mh105, 2021-12-19:
Thanks! @agramfort 

Since the published tutorial treats SNR in amplitude, would it make sense to update the implementation details white page to say the same? This could be helpful if people have a priori knowledge about SNR in their data - they can use it correctly to specify lambda2 and not accidentally square it. 

And I assume the \lambda ~ 1 / SNR^2 in this [docstring](https://mne.tools/stable/generated/mne.minimum_norm.estimate_snr.html?highlight=snr#mne-minimum-norm-estimate-snr) is just a typo then. It should say lambda2 \approx 1 / SNR^2 (SNR in amplitude).

PR #10172 (2022-01-03): MRG, DOC: Clarify lambda/SNR
Closes #10149

Make everything consistent by just talking about SNR of amplitudes, and correct an error in `estimate_snr`
closed 2022-01-03 by agramfort