mne-tools/mne-python #10161: Add pial.T1 brain format in mne.viz.brain or mne.viz.plot_alignment

BarryLiu97, 2021-12-27:
Hi, in Freesurfer7, the format of the surface file of pial seems to be __pial.T1__, and so far MNE seems does not support this.

agramfort, 2021-12-27:
can you give more details?

you run recon-all with what parameters and then when fails?

> Message ID: ***@***.***>
>

BarryLiu97, 2021-12-27:
Hi, the recon-all procedure was fine. But my Freesurfer7's surf result is different from Freesurfer6. To be specific, the lh.pial becomes lh.pial.T1. Also, lh.pial still exists in the surf file, but nibabel says it's not a surface file. I successfully load the lhpial.T1 using nibabel.
And API like mne.viz.plot_alignment doesn't accept 'pial.T1' as surf parameter's input.
My recon-all is
```
recon-all -i $T1 -s $SUBJECT -sd $SUBJECTs_DIR -all -qcache -cw256
```

agramfort, 2021-12-29:
ok I just tested locally. It's not that the file is not there on
freesurfer7 it is that you should have a symbolic link:

○ → ll sample/surf/lh.pial
lrwxrwxrwx 1 agramfor parietal 10 Dec 28 19:12 sample/surf/lh.pial ->
lh.pial.T1

did you move the file for a unix system to windows and therefore the link
broke?

> Message ID: ***@***.***>
>

BarryLiu97, 2021-12-29:
Hi, you mean in your case, lh.pial and lh.pial.T1 both exists. And lh.pial is a symbolic link to lh.pial.T1, which means it should both be OK if lh.pial or lh.pial.T1 loaded?
I did copy the result from ubuntu20 to WIndows10. I will try to open the lh.pial in the original computer to see if it's OK.

agramfort, 2021-12-29:
>
> Hi, you mean in your case, lh.pial and lh.pial.T1 both exist. And lh.pial
> is a symbolic link to lh.pial.T1,
>

yes

> which means it should both be OK if lh.pial or lh.pial.T1 loaded?
>

yes they are the same content

> I did copy the result from ubuntu20 to WIndows10. I will try to open the
> lh.pial in the original computer to see if it's OK.Message ID:
> ***@***.***>
>
the fix is to copy lh.pial.T1 to lh.pial and same for rh

BarryLiu97, 2021-12-29:
> Hi, you mean in your case, lh.pial and lh.pial.T1 both exist. And lh.pial is a symbolic link to lh.pial.T1,
> yes
> which means it should both be OK if lh.pial or lh.pial.T1 loaded?
> yes they are the [...]

Thank you so much for your help!

closed 2021-12-30 by agramfort