mne-tools/mne-python #10164: report.add_figure complains about bad tags

jasmainak, 2021-12-29:
#### Describe the bug
The report class complains about a bad tag but does not say what characters are bad. Furthermore, I think it should allow numeric characters. I get:

```py
ValueError: The following tags contained invalid characters: 'Subject 2002'
```

#### Steps to reproduce
Just do:

```py
import matplotlib.pyplot as plt
from mne.report import Report

rep = Report()
subject = '2002'
fig = plt.figure()
rep.add_figure(fig, title=f'Figure {i}', tags=(f'Subject {subject}',))
```

#### Expected results
Report is produced with tag "subject 2002"

#### Actual results
Numeric subjects are quite common and I'd imagine a common usecase would be to use numeric tags. I think they should work ... did work previously in `rep.add_figs_to_section`, so I'd consider this a regression.

#### Additional information
I'm on latest master branch

jasmainak, 2021-12-29:
Also the docstring talks about "collection of str". I thought "collection" was a matplotlib object? It actually crosslinks to matplotlib. Is that what we mean? Shouldn't it be `tuple | list` or something like that?

jasmainak, 2021-12-29:
Another issue is that the `rep.add_figure` closes the figure. I don't think this should happen as user may still like to interact with the matplotlib figure while having it in a report for easier digestion.

agramfort, 2021-12-30:
can you give a try to the fix?

> Message ID: ***@***.***>
>

hoechenberger, 2022-01-03:
> Also the docstring talks about "collection of str". I thought "collection" was a matplotlib object? It actually crosslinks to matplotlib. Is that what we mean?

We mean a Python [collection](https://docs.python.org/3/library/collections.html) / container type, i.e., `list`, `set`, `tuple`, `array`. In the dev docstring, this has already been changed to `array-like`: https://mne.tools/dev/generated/mne.Report.html

However we should adjust the error message as well.

We don't allow for [certain characters](https://github.com/mne-tools/mne-python/blob/fecf5411d28f47dcb4ff379976a1b4813edc45a9/mne/report/report.py#L304) in tags, including spaces.

[`_fig_to_img()`](https://github.com/mne-tools/mne-python/blob/fecf5411d28f47dcb4ff379976a1b4813edc45a9/mne/report/report.py#L344) has an `auto_close` parameter (defaults to `True`). We could expose this via the `add_figure()` API. WDYT?

jasmainak, 2022-01-03:
no strong feelings about exposing `auto_close`. I think if MNE-report creates a figure, it should auto-close, but if user created the figure (as in `add_figure`), it should not close, by default.

hoechenberger, 2022-01-03:
@jasmainak I think this makes sense, yes!

PR #10183 (2022-01-22): MRG: Better error message for bad tags in Report, and allow tags to be a string
Fixes #10164

@jasmainak would this already improve things for you a bit?
commit ebaaed43 (2022-01-22): MRG: Better error message for bad tags in Report, and allow tags to be a string (#10183)

* Better error message for bad tags

Fixes #10164

* Style

* Address review comments
closed 2022-01-22 by agramfort