mne-tools/mne-python #10165: Conda solver fails when installed with Pytorch + Sci-kit

FuriouslyCurious, 2021-12-30:
### Issue

When I try to install pytorch, sci-kit family of packages, and mne at the same time, conda solver is not able to work through all the dependencies and gets stuck in a recursive solver loop with following error message:

```
#8 86.88 Collecting package metadata (repodata.json): ...working... done
#8 154.0 Solving environment: ...working... failed with initial frozen solve. Retrying with flexible solve.
Examining scikit-image:  18%|█▊        | 10/56 [00:20<02:03,  2.69s/it]
```

### Potential Resolution

Potential resolution would be to revise requirements.txt / .yml to be less restrictive.  This would allow conda solver to find a solution.

### Temporary Workaround

Install MNE and MNE-NIRS with pip instead of conda

cbrnr, 2021-12-30:
I personally recommend `pip`, but that's not our official position. If you want to use conda, try mamba – it is *much* faster and *much* better in resolving dependencies than conda.

drammock, 2021-12-30:
I can confirm that with [mamba](https://mamba.readthedocs.io/en/latest/installation.html) the following resolves in about 15 seconds and installation completes in under 90 seconds:

```sh
mamba create -n temp
conda activate temp
mamba install mne scikit-image pytorch  # add "-c conda-forge" if it's not your default
```

FuriouslyCurious, 2021-12-30:
Thanks - I will check out mamba!

agramfort, 2021-12-30:
if mne-nirs pins with == a version of pandas I suspect it's the root cause.
Pinning versions with == and not >= makes easily environnements unsolvable.

> Message ID: ***@***.***>
>

drammock, 2021-12-30:
> if mne-nirs pins with == a version of pandas I suspect it's the root cause. Pinning versions with == and not >= makes easily environnements unsolvable.

Maybe so, but this issue is about MNE-Python (not MNE-nirs) and we don't pin pandas at all (not even a greater-than).

FuriouslyCurious, 2021-12-30:
Hi @drammock , 
I noticed same as @agramfort that requirements.txt for mne-nirs is pinning Pandas with "==".  

I also tried installing with mamba, and the install failed due to mayavi dependency:

```
#11 27.28 Channel: pytorch[linux-64,noarch], platform: linux-64, prio: 2 : 0
#11 27.28 Cache path:  /me/pkgs/cache/88c5dae4.json
#11 27.28 Channel: pytorch[linux-64,noarch], platform: noarch, prio: 2 : 0
#11 27.28 Cache path:  /me/pkgs/cache/c7c28c8a.json
#11 27.28 Channel: conda-forge[linux-64,noarch], platform: linux-64, prio: 1 : 0
#11 27.28 Cache path:  /me/pkgs/cache/b556ea5a.json
#11 27.28 Channel: conda-forge[linux-64,noarch], platform: noarch, prio: 1 : 0
#11 27.28 Cache path:  /me/pkgs/cache/ddfd5f96.json
#11 27.28
#11 27.28 Pinned packages:
#11 27.28   - python 3.9.*
#11 27.28
#11 27.28
#11 27.28 Encountered problems while solving:
#11 27.28   - package mne-0.19.2-py_2 requires mayavi, but none of the providers can be installed
```

agramfort, 2021-12-30:
0.19 is super old.

latest release is 0.24.1 see https://pypi.org/project/mne/

it's available on conda forge: https://anaconda.org/conda-forge/mne

> Message ID: ***@***.***>
>

FuriouslyCurious, 2021-12-30:
Interesting - I have only two channels in my priority list (pytorch, conda-forge) and still the solver seems to be picking up a much older version. 

I will try to force it to pick up a higher version.

drammock, 2021-12-30:
> I also tried installing with mamba, and the install failed due to mayavi dependency:

If you post any more follow-ups, please include the command that led to the output, not just the output.

FuriouslyCurious, 2021-12-30:
@drammock  of course - pasting the commands below.  

**All commands are run on a fresh VM instance.** 
Commands to reproduce:
```
curl -L -O https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-$(uname)-$(uname -m).sh
bash Mambaforge-$(uname)-$(uname -m).sh
conda config --set channel_priority strict
conda config --add channels pytorch conda-forge
mamba install -vy jupyterlab ipywidgets ipympl nodejs rich arrow \
                      pytorch torchvision torchaudio cudatoolkit=11.3 \
                      scikit-learn scikit-image \
                      pillow imageio 

* SUCCESS up to this ^ point 
```

### Attempt 1:
Running mne install with no pinned version picks up mne 0.19:
```
mamba install -vy mne mne-nirs

* PIcks MNE 0.19
```

### Attempt 2
Trying with pinned mne version gives following error:
```
mamba install -vy mne=0.24.1 mne-nirs=0.1.2
#12 15.32 info     Parsing MatchSpec mne=0.24.1
#12 15.32 info     Parsing MatchSpec mne=0.24.1
#12 15.32 info     Adding job: mne =0.24.1*
#12 15.32 info     Parsing MatchSpec mne-nirs=0.1.2
#12 15.32 info     Parsing MatchSpec mne-nirs=0.1.2
#12 15.32 info     Adding job: mne-nirs =0.1.2*
#12 15.32 info     Parsing MatchSpec ca-certificates
#12 15.32 info     Parsing MatchSpec ca-certificates
#12 15.32 info     Parsing MatchSpec certifi
#12 15.32 info     Parsing MatchSpec certifi
#12 15.32 info     Parsing MatchSpec openssl
#12 15.32 info     Parsing MatchSpec openssl
#12 15.32 info     Parsing MatchSpec python 3.9.*
#12 18.40 info     Problem count: 1
#12 18.40 info     Freeing pool.
#12 18.41 info     Freeing solver.
#12 18.42
#12 18.42                   __    __    __    __
#12 18.42                  /  \  /  \  /  \  /  \
#12 18.42                 /    \/    \/    \/    \
#12 18.42 ███████████████/  /██/  /██/  /██/  /████████████████████████
#12 18.42               /  / \   / \   / \   / \  \____
#12 18.42              /  /   \_/   \_/   \_/   \    o \__,
#12 18.42             / _/                       \_____/  `
#12 18.42             |/
#12 18.42         ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
#12 18.42         ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
#12 18.42         ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
#12 18.42         ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
#12 18.42         ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
#12 18.42         ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝
#12 18.42
#12 18.42         mamba (0.19.1) supported by @QuantStack
#12 18.42
#12 18.42         GitHub:  https://github.com/mamba-org/mamba
#12 18.42         Twitter: https://twitter.com/QuantStack
#12 18.42
#12 18.42 █████████████████████████████████████████████████████████████
#12 18.42
#12 18.42
#12 18.42 Looking for: ['mne=0.24.1', 'mne-nirs=0.1.2']
#12 18.42
#12 18.42 Channel: pytorch[linux-64,noarch], platform: linux-64, prio: 2 : 0
#12 18.42 Cache path:  /pkgs/cache/88c5dae4.json
#12 18.42 Channel: pytorch[linux-64,noarch], platform: noarch, prio: 2 : 0
#12 18.42 Cache path:  /pkgs/cache/c7c28c8a.json
#12 18.42 Channel: conda-forge[linux-64,noarch], platform: linux-64, prio: 1 : 0
#12 18.42 Cache path:  /pkgs/cache/b556ea5a.json
#12 18.42 Channel: conda-forge[linux-64,noarch], platform: noarch, prio: 1 : 0
#12 18.42 Cache path:  /pkgs/cache/ddfd5f96.json
#12 18.42
#12 18.42 Pinned packages:
#12 18.42   - python 3.9.*
#12 18.42
#12 18.42
#12 18.42 Encountered problems while solving:
#12 18.42   - package mne-0.24.1-hd8ed1ab_0 requires vtk >=9.0.1, but none of the providers can be installed
#12 18.42
#12 18.42 pytorch/noarch
#12 18.42 pytorch/linux-64
#12 18.42 conda-forge/noarch
#12 18.42 conda-forge/linux-64
```

### Attempt 3
I tried adding vtk to the install command, then it gives ffmpeg error.  If I add ffmpeg, it gives another error.
```
mamba install -vy mne=0.24.1 mne-nirs=0.1.2 vtk ffmpeg

#12 22.04 Pinned packages:
#12 22.04   - python 3.9.*
#12 22.04
#12 22.04
#12 22.04 Encountered problems while solving:
#12 22.04   - package vtk-9.0.1-egl_py39h4ab89ca_10 requires ffmpeg >=4.3.2,<5.0a0, but none of the providers can be installed
#12 22.04
```

drammock, 2021-12-30:
You will have better success debugging conda install issues like this if you install everything at once.  If I add `mne` and `mne-nirs` to your big install line, what I see is this:

```console
$ mamba install mne mne-nirs jupyterlab ipywidgets ipympl nodejs rich arrow pytorch torchvision torchaudio cudatoolkit=11.3 scikit-image imageio pillow

[mamba logo suppressed]

Looking for: ['mne', 'mne-nirs', 'jupyterlab', 'ipywidgets', 'ipympl', 'nodejs', 'rich', 'arrow', 'pytorch', 'torchvision', 'torchaudio', 'cudatoolkit=11.3', 'scikit-image', 'imageio', 'pillow']

conda-forge/linux-64     Using cache
conda-forge/noarch       Using cache
numba/noarch             [====================] (00m:00s) No change
pytorch/linux-64         [====================] (00m:00s) No change
numba/linux-64           [====================] (00m:00s) No change
pytorch/noarch           [====================] (00m:00s) No change
Encountered problems while solving:
  - package cudatoolkit-11.3.1-ha36c431_9 has constraint __cuda >=11.3.1 conflicting with __cuda-11.2-0
```

So you'll need to figure out which packages are constraining cuda version (presumably one of them is an `mne` dependency) or else relax your constraint on `cudatoolkit` (then it works, but installs cudatoolkit 11.1.1).

rob-luke, 2021-12-30:
> I noticed same as @agramfort that requirements.txt for mne-nirs is pinning Pandas with "==".

Can you open a PR in MNE-NIRS to fix this?

FuriouslyCurious, 2021-12-31:
> > I noticed same as @agramfort that requirements.txt for mne-nirs is pinning Pandas with "==".
> 
> Can you open a PR in MNE-NIRS to fix this?

Done - submitted a pull req as well.

larsoner, 2022-01-04:
Closing here for https://github.com/mne-tools/mne-nirs/pull/426

closed 2022-01-04 by larsoner