mne-tools/mne-python #10167: Wrong annotation when export eeg to edf after cropping and set_annotation

BarryLiu97, 2022-01-02:
Here is a sample code"
```
import mne
fpath = ...
raw = mne.io.read_raw_fif(fpath, preload=True)
annot = mne.Annotations(...)
raw.set_annotations(onset=[15], duration=[0], description=['start'])

raw.crop(10, 100)
raw.export('raw.edf')
```
The bug is:
If the ```start``` label is at 15 seconds in the original file, if I crop the data from 10 to 100 seconds, so the ```start``` label should be at 5 seconds. It's OK if I saved the file in FIF format, but to EDF format, the ```start``` label is at 10 + 15 seconds.

agramfort, 2022-01-03:
#10152 may have fixed it. Can you check?

BarryLiu97, 2022-01-03:
If I want to try that, does that mean I need to install the dev version?

agramfort, 2022-01-03:
yes

> Message ID: ***@***.***>
>

larsoner, 2022-01-04:
This works for me on `main`, so I think #10152 did fix it:
```
import mne
import numpy as np
data = np.random.RandomState(0).randn(1, 100001)
info = mne.create_info(1, 1000., 'eeg')
raw = mne.io.RawArray(data, info)
annot = mne.Annotations(onset=[15], duration=[0], description=['start'])
raw.set_annotations(annot)
raw.crop(10, 100)
raw.export('raw.edf')
```
@BarryLiu97 feel free to reopen if you see otherwise on your data

closed 2022-01-04 by larsoner