mne-tools/mne-python #10173: can't plot using Brain class

drammock, 2022-01-03:
```py
In [1]: import mne
In [2]: Brain = mne.viz.get_brain_class()
In [3]: brain = Brain('sample')
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-3-7cf1a9f56a3b> in <module>
----> 1 brain = Brain('sample')

/opt/mne/python/mne/viz/_brain/_brain.py in __init__(***failed resolving arguments***)
    525         logger.debug(f'Hemi offset: {offset}')
    526 
--> 527         self._renderer = _get_renderer(name=self._title, size=size,
    528                                        bgcolor=self._bg_color,
    529                                        shape=shape,

/opt/mne/python/mne/viz/backends/renderer.py in _get_renderer(*args, **kwargs)
     37 
     38 def _get_renderer(*args, **kwargs):
---> 39     _get_3d_backend()
     40     return backend._Renderer(*args, **kwargs)
     41 

/opt/mne/python/mne/viz/backends/renderer.py in _get_3d_backend()
    163             for name in VALID_3D_BACKENDS:
    164                 try:
--> 165                     _reload_backend(name)
    166                 except ImportError as exc:
    167                     errors[name] = str(exc)

/opt/mne/python/mne/viz/backends/renderer.py in _reload_backend(backend_name)
     31 def _reload_backend(backend_name):
     32     global backend
---> 33     backend = importlib.import_module(name=_backend_name_map[backend_name],
     34                                       package='mne.viz.backends')
     35     logger.info('Using %s 3d backend.\n' % backend_name)

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/__init__.py in import_module(name, package)
    125                 break
    126             level += 1
--> 127     return _bootstrap._gcd_import(name[level:], package, level)
    128 
    129 

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/_bootstrap.py in _gcd_import(name, package, level)

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/_bootstrap.py in _find_and_load(name, import_)

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/_bootstrap.py in _load_unlocked(spec)

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/_bootstrap_external.py in exec_module(self, module)

/opt/miniforge3/envs/mnedev/lib/python3.9/importlib/_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

/opt/mne/python/mne/viz/backends/_qt.py in <module>
      9 
     10 import pyvista
---> 11 from pyvistaqt.plotting import FileDialog
     12 
     13 from PyQt5.QtCore import Qt, pyqtSignal, QLocale

/opt/miniforge3/envs/mnedev/lib/python3.9/site-packages/pyvistaqt/__init__.py in <module>
      1 """PyVista package for 3D plotting and mesh analysis."""
      2 from pyvistaqt._version import __version__
----> 3 from pyvistaqt.plotting import BackgroundPlotter, MainWindow, MultiPlotter, QtInteractor
      4 
      5 __all__ = [

/opt/miniforge3/envs/mnedev/lib/python3.9/site-packages/pyvistaqt/plotting.py in <module>
     54 from pyvista.plotting.plotting import BasePlotter
     55 from pyvista.utilities import conditional_decorator, threaded
---> 56 from qtpy import QtCore
     57 from qtpy.QtCore import QSize, QTimer, Signal
     58 from qtpy.QtWidgets import (

/opt/miniforge3/envs/mnedev/lib/python3.9/site-packages/qtpy/__init__.py in <module>
    110 API = os.environ[QT_API].lower()
    111 initial_api = API
--> 112 assert API in (PYQT5_API + PYQT4_API + PYSIDE_API + PYSIDE2_API)
    113 
    114 is_old_pyqt = is_pyqt46 = False

AssertionError: 
ipdb> p API
'pyqt6'
```

Oddly, `mamba list` shows this (excerpt):
```
# Name                    Version                   Build  Channel
ipyvtklink                0.2.1              pyhd8ed1ab_1    conda-forge
pyvista                   0.32.1             pyhd8ed1ab_0    conda-forge
pyvistaqt                 0.5.0              pyhd8ed1ab_0    conda-forge
```
but `mne.sys_info()` says pyvistaqt / ipyvtklink is not found:

```
pyvista:        0.32.1 {OpenGL 4.5.0 NVIDIA 460.91.03 via GeForce RTX 2060/PCIe/SSE2}
pyvistaqt:      Not found
ipyvtklink:     Not found
vtk:            9.0.1
PyQt5:          5.12.3
```

Things tried so far:

- upgrading to pyvistaqt 0.6.0

larsoner, 2022-01-03:
Do you also have PyQt6 installed? Does the env var `QT_API=pyqt5` fix it?

larsoner, 2022-01-03:
... another thing to try would be upgrading `qtpy`

drammock, 2022-01-03:
- I don't have pyqt6 installed (see below)
- using env variable does not fix it.
- upgrading qtpy (from 1.11.2 to 2.0.0) fixed it

```
$ mamba list pyqt
# Name                    Version                   Build  Channel
pyqt                      5.12.3           py39hf3d152e_8    conda-forge
pyqt-impl                 5.12.3           py39hde8b62d_8    conda-forge
pyqt5-sip                 4.19.18          py39he80948d_8    conda-forge
pyqtchart                 5.12             py39h0fcd23e_8    conda-forge
pyqtgraph                 0.12.3             pyhd8ed1ab_0    conda-forge
pyqtwebengine             5.12.1           py39h0fcd23e_8    conda-forge
```

closed 2022-01-03 by drammock