mne-tools/mne-python #10180: Add Maximum Intensities Projection mode in mne.gui.locate_ieeg

BarryLiu97, 2022-01-05:
#### Describe the new feature or enhancement
In Brainstorm, when locating iEEG, sometimes a MIP mode is pretty efficient. Consider adding a MIP mode to the locate_ieeg GUI.

#### Describe your proposed implementation
Might just add a new QPushButton with __Maximum Intensities Projection__ text.

#### Additional comments
The MIP mode of CT looks like this:
```
import nibabel as nib

ct_path = '...'
ct = nib.load(ct_path)

ct_data = np.asarray(ct.dataobj)

sagittal_mip = np.amax(ct_data, axis=0)
coronal_mip = np.amax(ct_data, axis=2)
axial_mip = np.amax(ct_data, axis=1)

fig, ax = plt.subplots(2, 2, dpi=200)
ax[0, 0].imshow(coronal_mip, cmap='Greys_r')
ax[0, 0].set_title('Coronal', size=9)
ax[0, 0].axis('off')
ax[0, 1].imshow(sagittal_mip, cmap='Greys_r')
ax[0, 1].set_title('Sagittal', size=9)
ax[0, 1].axis('off')
ax[1, 0].imshow(axial_mip, cmap='Greys_r')
ax[1, 0].set_title('Axial', size=9)
ax[1, 0].axis('off')
ax[1, 1].axis('off')
plt.show(block=True)
```
[image]
Sorry, I don't know how to transfer them in the right direction.

alexrockhill, 2022-01-05:
Looks very cool and useful! I'll have to think about the best way to display it though, it might be better as a popup because it might not make sense to navigate when you can't see where you're going. Maybe it could be displayed normally but we turn off navigation as another option. Or just leave it on and hope people won't get confused. What do you think @BarryLiu97?

BarryLiu97, 2022-01-05:
I agree with 
> display normally but we turn off navigation as another option.
I was thinking adding a Label and a Button next to __show brain__
![1641416969(1)](https://user-images.githubusercontent.com/65115609/148289867-50b37a88-de1f-4a4c-9d40-
d75d646939d8.jpg)
@alexrockhill

closed 2022-01-08 by agramfort