mne-tools/mne-python #10190: Make _hid_match() public?

hoechenberger, 2022-01-08:
In the MNE-BIDS-Pipeline, I [need](https://github.com/mne-tools/mne-bids-pipeline/pull/503) the functionality of `mne.utils.mixin._hid_match()`; however, the function is not part of the public API.

Is there any chance we could make it public?

cc @agramfort

agramfort, 2022-01-09:
use a hack like this for now:

if would use here something like this for now:

```
import numpy as np
import mne

event_id = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
            'visual/right': 4}

info = mne.create_info(['A'], 1)
data = np.random.randn(len(event_id), 1, 1)
events = np.zeros((len(event_id), 3), dtype=int)
events[:, 0] = events[:, 2] = np.arange(len(event_id)) + 1
epochs = mne.EpochsArray(data, info, event_id=event_id, events=events)
event_id_hid = epochs['left']
print(event_id_hid.event_id)
```

what we call HID here is a cryptic name. If we expose such en event_id filtering function we need to find a good naming.

hoechenberger, 2022-01-09:
@agramfort Thanks for providing this example code!

Alas, the reason why I didn't implement that myself (you had suggested it earlier) was that I wasn't really convinced it's what we'd want in the BIDS Pipeline. If this were just a part of some other package – alright. But for the BIDS Pipeline, I'd like to keep the code readable and free of hacks as good as we can … and this here is clearly a (granted, clever!) hack 😅

But if it's really just about the name of the function, we should try to come up with something better / more descriptive, for example, `resolve_event_names()`, `find_event_names()`, `events_in_mapping()`, …?

agramfort, 2022-01-11:
ok I slept over it. How about a public function called

`mne.utils.filter_event_id(event_id, query: str)`

or something like this? I would like to avoid the use of HID in the public API.

agramfort, 2022-01-11:
query or keys I don't know...

hoechenberger, 2022-01-11:
I like the idea.

We could make it such that the function accepts as its first parameter a dict (our good old `event_id`) or a collection of strings (basically, `event_id.keys()`).

I think `key` is good as a name for the second parameter.

agramfort, 2022-01-12:
works for me

> Message ID: ***@***.***>
>

PR #10233 (2022-01-22): MRG: Add match_event_names()
This basically publicly exposes the behavior of the previously-private function `_hid_match()`.

Fixes #10190

I went with a slightly different name than originally discussed.
commit e305764e (2022-01-22): MRG: Add match_event_names() (#10233)

* Add match_event_names()

Fixes #10190

* Update changelog

* Update mne/event.py

* Line too long

* Improve tests

* Fix doctest?

* Remove cruft [ci skip]

* Update mne/event.py

* Move example [skip azp][skip actions]
closed 2022-01-22 by larsoner