mne-tools/mne-python #10193: Docker-ized MNE not detecting PyVista and Mayavi backends

FuriouslyCurious, 2022-01-10:
**Steps to reproduce:**

Install MNE inside a docker image:

```
mamba install -c conda-forge -vy jupyterlab ipywidgets ipympl nodejs pytorch torchvision mne mne-nirs

conda list shows tht Mayavi and PyVista both are installed.  Output edited for brevity.
conda list                                                                                   (base)
# packages in environment:
#
# Name                    Version                   Build  Channel

ipycanvas                 0.10.2             pyhd8ed1ab_0    conda-forge
ipyevents                 2.0.1              pyhd8ed1ab_0    conda-forge
ipykernel                 6.6.1            py39hef51801_0    conda-forge
ipympl                    0.8.5              pyhd8ed1ab_0    conda-forge
ipython                   7.31.0           py39hf3d152e_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
ipyvtklink                0.2.1              pyhd8ed1ab_1    conda-forge
ipywidgets                7.6.5              pyhd8ed1ab_0    conda-forge

jupyterlab                3.2.6              pyhd8ed1ab_0    conda-forge
jupyterlab_pygments       0.1.2              pyh9f0ad1d_0    conda-forge
jupyterlab_server         2.10.3             pyhd8ed1ab_0    conda-forge
jupyterlab_widgets        1.0.2              pyhd8ed1ab_0    conda-forge

mamba                     0.19.1           py39hfa8f2c8_0    conda-forge
matplotlib                3.5.1            py39hf3d152e_0    conda-forge
matplotlib-base           3.5.1            py39h2fa2bec_0    conda-forge
matplotlib-inline         0.1.3              pyhd8ed1ab_0    conda-forge
mayavi                    4.7.2            py39h71d8d94_5    conda-forge

mne                       0.24.1               hd8ed1ab_1    conda-forge
mne-base                  0.24.1             pyhd8ed1ab_1    conda-forge
mne-nirs                  0.1.2              pyhd8ed1ab_0    conda-forge
mne-qt-browser            0.1.7              pyhd8ed1ab_0    conda-forge

nibabel                   3.2.1              pyhd8ed1ab_0    conda-forge
nilearn                   0.8.1              pyhd8ed1ab_0    conda-forge
ninja                     1.10.2               h4bd325d_1    conda-forge
nodejs                    16.12.0              h92b4a50_0    conda-forge
notebook                  6.4.6              pyha770c72_0    conda-forge

numba                     0.53.1           py39h56b8d98_1    conda-forge
numexpr                   2.7.3            py39hde0f152_1    conda-forge
numpy                     1.22.0           py39h91f2184_0    conda-forge

pillow                    8.4.0            py39ha612740_0    conda-forge
pip                       21.3.1             pyhd8ed1ab_0    conda-forge
pooch                     1.5.2              pyhd8ed1ab_0    conda-forge
pyqt                      5.12.3           py39hf3d152e_8    conda-forge
pyqt-impl                 5.12.3           py39hde8b62d_8    conda-forge
pyqt5-sip                 4.19.18          py39he80948d_8    conda-forge
pyqtchart                 5.12             py39h0fcd23e_8    conda-forge
pyqtgraph                 0.12.3             pyhd8ed1ab_0    conda-forge
pyqtwebengine             5.12.1           py39h0fcd23e_8    conda-forge
python                    3.9.9           h62f1059_0_cpython    conda-forge
pytorch                   1.9.0           cpu_py39hfbcbfe4_0    conda-forge
pytorch-cpu               1.9.0           cpu_py39h718b53a_0    conda-forge
pyvista                   0.33.0             pyhd8ed1ab_0    conda-forge
pyvistaqt                 0.6.0              pyhd8ed1ab_1    conda-forge
qt                        5.12.9               hda022c4_4    conda-forge
qtpy                      2.0.0              pyhd8ed1ab_0    conda-forge

```

**Problem:**

MNE sys info doesn't recognize Mayavi or PyVista.  Provides the same results in REPL as well as Jupyter Lab Notebook.
```

echo -e "import mne; mne.sys_info()" | exec python3
Platform:       Linux-5.10.76-linuxkit-x86_64-with-glibc2.31
Python:         3.9.9 | packaged by conda-forge | (main, Dec 20 2021, 02:41:03)  [GCC 9.4.0]

mne:            0.24.1
numpy:          1.22.0 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:          1.7.3
matplotlib:     3.5.1 {backend=agg}
sklearn:        1.0.2
numba:          0.53.1
nibabel:        3.2.1
nilearn:        0.8.1
dipy:           1.4.1
cupy:           Not found
pandas:         1.3.5
mayavi:         Not found
pyvista:        Not found
pyvistaqt:      Not found
ipyvtklink:     0.2.1
vtk:            Not found
PyQt5:          5.12.3
ipympl:         0.8.5
mne_qt_browser: 0.1.7
pooch:          v1.5.2

```

**Trying to set "Notebook" (or any other) 3D backend:**
Provides the same results in REPL as well as Jupyter Lab Notebook.

```
echo -e "import mne; mne.viz.set_3d_backend(\"notebook\")" | exec python3
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<decorator-gen-182>", line 24, in set_3d_backend
  File "/lib/python3.9/site-packages/mne/viz/backends/renderer.py", line 129, in set_3d_backend
    _reload_backend(backend_name)
  File "/lib/python3.9/site-packages/mne/viz/backends/renderer.py", line 34, in _reload_backend
    backend = importlib.import_module(name=_backend_name_map[backend_name],
  File "/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/lib/python3.9/site-packages/mne/viz/backends/_notebook.py", line 10, in <module>
    import pyvista
  File "/lib/python3.9/site-packages/pyvista/__init__.py", line 12, in <module>
    from pyvista.plotting import *
  File "/lib/python3.9/site-packages/pyvista/plotting/__init__.py", line 4, in <module>
    from .charts import Chart2D, ChartMPL, ChartBox, ChartPie
  File "/lib/python3.9/site-packages/pyvista/plotting/charts.py", line 13, in <module>
    from pyvista import _vtk
  File "/lib/python3.9/site-packages/pyvista/_vtk.py", line 369, in <module>
    import vtkmodules.vtkRenderingContextOpenGL2  # Necessary for displaying charts, otherwise crashes on rendering
ImportError: libGL.so.1: cannot open shared object file: No such file or directory

```

**Installing OpenGL libraries inside Docker:**

I installed all OpenGL libraries inside Docker (libgl1-mesa-glx, ffmpeg, libsm6, libxext6, python3-opencv), but the result remains the same.  MNE sys info does not detect these backends.

larsoner, 2022-01-10:
Until `python -c "import vtk"` works, MNE/PyVista will not work. If you do `glxinfo | grep profile` in your dockerized environment, does it work? You might need to `sudo apt install mesa-utils` to get it.

larsoner, 2022-01-10:
... looking at our CircleCI config:

https://github.com/mne-tools/mne-python/blob/main/.circleci/config.yml#L95

You probably need at least `libosmesa6 libglx-mesa0 libopengl0 libglx0` -- I assume some of these were not installed by `libgl1-mesa-glx`...?

bloyl, 2022-01-11:
Might be worth looking at https://github.com/mne-tools/mne-docker
that has docker files that are close to what I'm actively using.

larsoner, 2022-03-09:
Closing since I think there are solutions in here that should work, but let's reopen if none of them do

closed 2022-03-09 by larsoner