mne-tools/mne-python #10195: Annotation error after mne.io.concatenate_raws() and raw.crop()

kcsaba12, 2022-01-12:
#### Describe the bug
If two raw files with measure date are concatenated then after the crop function the annotations are mixed between the two files. 
See first attached picture: https://imgur.com/a/BEHAZLh

If no measure date are set (None) the annotations are proper: https://imgur.com/a/HqnbGTL

This is a problem in case of real EEG files...

#### Steps to reproduce
```
from datetime import datetime, timezone

import mne
import numpy as np

CH_NUM = 32
FS = 160

ch_names = [f'EEG{i}' for i in range(CH_NUM)]
ch_types = ['eeg'] * len(ch_names)

def _create_raw(eeg, ch_names, ch_types, fs, onset, duration, description, rec_date=None):
    info = mne.create_info(ch_names, ch_types=ch_types, sfreq=fs)
    raw = mne.io.RawArray(eeg, info)
    annotation = mne.Annotations(onset, duration, description)
    raw = raw.set_annotations(annotation)
    if rec_date is not None:
        raw.set_meas_date(rec_date)
    return raw

data = np.array([list(range(40)) * 4 * 10] * CH_NUM) * 5 * 1e-7
onset = [2.5, 5, 6, 7, 8]
duration = [.1] * len(onset)
description = [12, 'on', 1, 2, 'off']
rec_date = datetime(2022, 1, 12, 10, 12, 5, tzinfo=timezone.utc)
raw1 = _create_raw(data, ch_names, ch_types, FS, onset, duration, description, rec_date)

data = np.array([([1] * int(FS / 2) + [0] * int(FS / 2)) * 10] * CH_NUM) * 1e-5
onset = [1.5, 2, 2.7, 5]
duration = [.1] * len(onset)
description = ['on', 3, 4, 'off']
rec_date = datetime(2022, 1, 12, 10, 15, 5, tzinfo=timezone.utc)
raw2 = _create_raw(data, ch_names, ch_types, FS, onset, duration, description, rec_date)

raw = mne.io.concatenate_raws([raw1, raw2])
start_mask = raw.annotations.description == 'on'
end_mask = raw.annotations.description == 'off'
tmins = raw.annotations.onset[start_mask]
tmaxs = raw.annotations.onset[end_mask]

for i, tmin in enumerate(tmins):
    tmax = tmaxs[i]
    sess = raw.copy()
    sess.crop(tmin, tmax)
    sess.plot(block=False)

raw.plot(block=True)
```

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
replace
`rec_date = datetime(...)`
with
`rec_date = None`

expected result: https://imgur.com/a/HqnbGTL

#### Additional information
Backend TkAgg is interactive backend. Turning interactive mode on.
Platform:       Windows-10-10.0.22000-SP0
Python:         3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)]
Executable:     C:\Programs\Miniconda3\envs\bci\python.exe
CPU:            Intel64 Family 6 Model 158 Stepping 10, GenuineIntel: 12 cores
Memory:         Unavailable (requires "psutil" package)
mne:            1.0.dev0
numpy:          1.19.5 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:          1.7.1
matplotlib:     3.4.3 {backend=TkAgg}
sklearn:        1.0.1
numba:          Not found
nibabel:        Not found
nilearn:        Not found
dipy:           Not found
cupy:           Not found
pandas:         1.3.4
pyvista:        Not found
pyvistaqt:      Not found
ipyvtklink:     Not found
vtk:            Not found
PyQt5:          Not found
ipympl:         Not found
mne_qt_browser: Not found
pooch:          Not found

agramfort, 2022-01-12:
I can replicate the pb. thanks @kcsaba12 for the detailed bug fix. We'll try to look asap.

kcsaba12, 2022-01-13:
@agramfort Your welcome! 

I found an additional bug which is the cause of solving the upper bug....
[image]

- Left image: bug visualization - When 2 files are concatenated which does not have measurement day the labels are shifted.
- Right image: expected behaviour

**Steps to reproduce**
```
from datetime import datetime, timezone

import mne
import numpy as np

CH_NUM = 32
FS = 160

ch_names = [f'EEG{i}' for i in range(CH_NUM)]
ch_types = ['eeg'] * len(ch_names)

def _create_raw(eeg, ch_names, ch_types, fs, onset, duration, description, rec_date=None):
    info = mne.create_info(ch_names, ch_types=ch_types, sfreq=fs)
    raw = mne.io.RawArray(eeg, info)
    annotation = mne.Annotations(onset, duration, description)
    raw = raw.set_annotations(annotation)
    if rec_date is not None:
        raw.set_meas_date(rec_date)
    return raw

data = np.array([list(range(40)) * 4 * 10] * CH_NUM) * 5 * 1e-7
onset = [2.5, 5, 6, 7, 8]
duration = [.1] * len(onset)
description = [12, 'on', 1, 2, 'off']
rec_date = datetime(2022, 1, 12, 10, 12, 5, tzinfo=timezone.utc)
raw1 = _create_raw(data, ch_names, ch_types, FS, onset, duration, description, rec_date)

data = np.array([([1] * int(FS / 2) + [0] * int(FS / 2)) * 10] * CH_NUM) * 1e-5
onset = [1.5, 2, 2.7, 5]
duration = [.1] * len(onset)
description = ['on', 3, 4, 'off']
rec_date = datetime(2022, 1, 12, 10, 15, 5, tzinfo=timezone.utc)
raw2 = _create_raw(data, ch_names, ch_types, FS, onset, duration, description, rec_date)

raw = mne.io.concatenate_raws([raw1, raw2])
# first bug fixed:
raw.set_meas_date(None)

start_mask = raw.annotations.description == 'on'
end_mask = raw.annotations.description == 'off'
tmins = raw.annotations.onset[start_mask]
tmaxs = raw.annotations.onset[end_mask]

files = []
for i, tmin in enumerate(tmins):
    tmax = tmaxs[i]
    sess = raw.copy()
    sess.crop(tmin, tmax)
    sess.plot(block=False)

    files.append(f'test_file_{i}_raw.fif')
    sess.save(files[i], overwrite=True)

raw.plot(block=True)

# second bug:
raw = mne.io.concatenate_raws([mne.io.read_raw(file) for file in files])
raw.plot(title='Labels are shifted if all meas_date=None')

# corrected:
raws = [mne.io.read_raw(file) for file in files]
raws[0].set_meas_date(datetime.now(tz=timezone.utc))
mne.io.concatenate_raws(raws).plot(title='Expected labels', block=True)
```

jsosulski, 2022-01-28:
I just ran into this bug as well. I was trying to publish an anonymized EEG dataset recorded in our lab, so I removed all timestamp information and noticed my whole analysis pipelines failing on the data without timestamps.

My current hotfix is to set an arbitrary date. Even the same constant timestamp for all raw files seems to work.

agramfort, 2022-01-29:
we'll fix it soon. Now for anonymized data what is recommended by BIDS is
to shift dates in the entire study by a random number of days.
It has some advantages eg keep in the order of acquisitions, time of the
day etc.

> Message ID: ***@***.***>
>

PR #10358 (2022-02-18): BUG: Fix bug with crop and annot
Closes #10195

This one took a long time, but hopefully with the added tests, `logger.debug` calls, and debugging code in the tests (in case something breaks) things will be easier to fix next time we have problems with Annotations.

@jsosulski @kcsaba12 feel free to test this branch. I think it makes the example from #10195 work as expected
closed 2022-02-18 by agramfort