mne-tools/mne-python #10197: Use packaging.version for comparing versions

cbrnr, 2022-01-12:
Because `distutils.Version` is deprecated, we have our own `_compare_version` function (https://github.com/mne-tools/mne-python/blob/main/mne/fixes.py#L32-L55). However, I was wondering (1) why this is in fixes and (2) why we do not use `packaging.version.Version` instead (which according to the `DeprecationWarning` is the recommended replacement). I'm happy to implement the changes (and probably move it back to `utils`?) if you think it makes sense. @hoechenberger WDYT?

hoechenberger, 2022-01-12:
Hello @cbrnr,
> However, I was wondering (1) why this is in fixes

because it "fixes" things for environments which don't have `setuptools` installed, which IIRC is currently not a dependency of MNE-Python.

> and (2) why we do not use `packaging.version.Version` instead (which according to the `DeprecationWarning` is the recommended replacement).

Because this would introduce a dependency on `packaging`, which seems less-widespread than `setuptools` :)

hoechenberger, 2022-01-12:
That said, I don't mind adding a few deps to make our code cleaner and leaner, but others may disagree

cbrnr, 2022-01-12:
`setuptools` is bootstrapped with every Python installation, as is `pip`. So I'm not sure this is really a fix when everybody has it.

hoechenberger, 2022-01-12:
> `setuptools` is bootstrapped with every Python installation, as is `pip`. 

At least not in `coda` environments; haven't tried with `pyenv` & `venv`, though.

cbrnr, 2022-01-12:
But yes, I understand that it's not part of the standard library, so if you compile Python yourself you don't have it.

cbrnr, 2022-01-12:
I'd make `setuptools` a dependency. It's used everywhere and it's the [recommended replacement of distlib](https://docs.python.org/3/library/distutils.html).

hoechenberger, 2022-01-12:
> At least not in `conda` environments; haven't tried with `pyenv` & `venv`, though.

Sorry, I stand corrected, I just tested this and indeed `conda` and `mamba` do install `setuptools` and `pip` along with `python`.

I think it's just `conda-forge` that handles things differently, and requires one to explicitly list `setuptools` as a runtime dependency if it's required.

hoechenberger, 2022-01-12:
> I'd make `setuptools` a dependency. It's used everywhere and it's the [recommended replacement of distlib](https://docs.python.org/3/library/distutils.html).

No objection from my end. As soon as `distutils.Version` gets removed, we'll have to depend on `setuptools` anyway.

cbrnr, 2022-01-12:
OK. But to summarize, we don't even need `setuptools` because the recommended package to do version checks is `packaging`, so I'd add that as a dependency. And then remove `_compare_version` from `fixes`.

hoechenberger, 2022-01-12:
> But to summarize, we don't even need `setuptools` because the recommended package to do version checks is `packaging`,

Yeah well no, I still think `setuptools` is the better choice in case we want to pick a dependency that "everybody already has installed"

Like I said I personally don't care, but other MNE devs will 😅

agramfort, 2022-01-12:
no opinion

> Message ID: ***@***.***>
>

cbrnr, 2022-01-12:
We coal so vendor it if we don't want an additional dependency.

larsoner, 2022-01-12:
+1 for packaging

> We could also vendor it if we don't want an additional dependency.

That might indeed make sense. But let's wait until we settle #10198 to do this (or not).

drammock, 2022-01-12:
> Then we could also remove the [corresponding warning filter](https://github.com/mne-tools/mne-python/commit/523576f6bd805dfbf3e882933d779c5a2fd6d740#diff-d6b8de46e1e16287a7eed50a9cb2ed209e1d62ec5257722a2c0fce32dd2c726a).

That warning filter is not for our code, it was added because a dependency (nilearn) hadn't switched off of LooseVersion yet (reported [here](https://github.com/nilearn/nilearn/issues/3108) and fixed upstream 6 days ago).  Whatever change you end up making, please add a comment to that line indicating why the warning filter exists (e.g., `# present in nilearn v 0.8.1, fixed in nilearn main`).  IMO all our warning filters should have this, so we know when to bother checking whether they're out of date

I'm also +1 for using `packaging` but I don't see the point of vendoring it. It's developed by pypa and "blessed" as the recommended replacement for this use case, and thus seems likely to be used widely / required by many other libraries / be present on the user's system already.

cbrnr, 2022-01-12:
Yes, we should require a comment for every new filter warning! Maybe someone has the time to check which of these filters are not even necessary anymore?

I'd also try to vendor as few packages as possible, especially when we're talking about core packages provided by pypa.

hoechenberger, 2022-01-12:
> I'd also try to vendor as few packages as possible, especially when we're talking about core packages provided by pypa.

+1

PR #10223 (2022-01-19): Use packaging.version in _compare_version
Fixes #10197. Do we want to move the function to a different module?
closed 2022-01-19 by hoechenberger