mne-tools/mne-python #10201: Pinch-zooming plot_alignment() figure only updates zoom status after an additional click

hoechenberger, 2022-01-13:
The user is left in the dark as to what the current zoom status is until they start interacting with the 3D head again. @GuillaumeFavelier I can show you this tomorrow maybe? Otherwise I'll extend this report to make it clearer what I mean.

This is an issue I've observed on macOS, so I'm not sure whether you'll see it on different operating systems.

GuillaumeFavelier, 2022-01-13:
yes let's take a look tomorrow, I'll try to reproduce on my linux too :+1:

hoechenberger, 2022-01-16:
Same issue is present in the coreg UI

Related upstream PR: https://github.com/pyvista/pyvistaqt/pull/143

GuillaumeFavelier, 2022-01-18:
@hoechenberger https://github.com/pyvista/pyvistaqt/pull/143 has been merged!
Feel free to try it again with the development version of `pyvistaqt` :+1:

hoechenberger, 2022-01-18:
It's super smooth now 🥳

closed 2022-01-18 by hoechenberger

hoechenberger, 2022-01-18:
@larsoner Any chance we might see a new release of pyvistaqt before we release MNE-Python 1.0?

larsoner, 2022-01-18:
@GuillaumeFavelier did the last release -- @GuillaumeFavelier are you up for doing another quick one?

GuillaumeFavelier, 2022-01-18:
`pyvistaqt 0.7.0` is getting deployed!