mne-tools/mne-python #10207: `__init__()` got an unexpected keyword argument `'extract_dir'`

PABannier, 2022-01-15:
When trying to get the data path of the `sample` dataset, a `TypeError` is thrown:
`TypeError: __init__() got an unexpected keyword argument 'extract_dir'`.

I run:

```
import mne
from mne.datasets import sample
data_path = sample.data_path()
```

Here's the full traceback:

[image]

#### Additional information
I tried the latest dev and stable versions. Same issue.

agramfort, 2022-01-15:
can you try to update pooch?

pip install -U pooch

Message ID: ***@***.***>
>

PABannier, 2022-01-15:
Works like a charm. 

Thanks @agramfort !

closed 2022-01-15 by PABannier