mne-tools/mne-python #10216: [ENH] Finite Element Modeling for Tissue Displacement by Intracranial Electrodes and Conductivity

alexrockhill, 2022-01-18:
This is a bit of a two-for-one but the last major things that I can think of that are specific to sEEG/ECoG which are not available currently in MNE are modeling tissue displacement from the electrodes and conductivity to determine which tissues contribute to a recording contact.

For the first one, the idea would be to have a mapping like the symmetric diffeomorpic registration that warped an image around contacts present in the CT and could be applied to any Freesurfer image (e.g. an  `aseg` file).

For the second one, I'm not as sure exactly how this would work depending on the referencing scheme, but my impression is that it would be similar to what is done in https://github.com/simnibs/simnibs.

For both, this would potentially mean adding an optional finite element library dependency, but this field seems to be mostly proprietary closed source although not all. The one that seems lightweight enough and promising based on their documentation to me is https://github.com/adtzlr/felupe. It's not that others like `dolfin` or https://sfepy.org/doc-devel/index.html aren't likely capable of doing this (probably with a few PRs), I like `felupe` because it doesn't have a text-based model input and deals directly with `numpy` arrays. I have one discussion started that I need to follow up on https://github.com/adtzlr/felupe/discussions/137, they have been super helpful in the one interaction.

These papers have methods descriptions that apply to the problem: https://doi.org/10.1371/journal.pone.0242704 and https://doi.org/10.1016/j.media.2019.06.004. These are great but it's a bit confusing to me that a pretty complex method would be shared entirely in descriptions of the algorithm with no code. I'm not sure how hard it would be to implement but it would have been a lot easier if it could just be translated rather than written from scratch. Maybe it would be worth asking the authors, it's probably written in MATLAB, but I doubt that would work well with relicensing.

I haven't done a ton of this, but it doesn't seem like so much computations to find these goals to a good approximation but all the solutions with FEM involve inverting a global stiffness matrix which is large and sparse and takes a lot of time and RAM. Possibly the element-free Galerkin method might be more computationally efficient with similar results. I think for a first pass it might be fine to have a reasonable approximation which can then be iterated on and tested. Since this isn't really the main focus of intracranial analysis, it would be really nice if it took under an hour to compute. I haven't gotten too far into figuring out what this would take to do (i.e. a couple-week project or full summer), these are just my thoughts from my first impression.

I think the main questions are:
- Is this the best way to go about determining which tissues are contributing to an intracranial recording?
- Is this something that is appropriate/not too big and too computationally intensive for testing to live in the main MNE-Python repository?
- Is there a dependency that someone is confident is the best way to implement FE things? And, relatedly, would this be incredibly onerous to implement in `numpy`/`scipy` so that's off the table?
- Is the element-free Galerkin method easy enough to code that it might be lower computation and thus fit the needs of MNE users better?

@larsoner mentioned that maybe @agramfort might have some interest in this potentially for reviewing some PRs.

larsoner, 2022-01-18:
@jasmainak you probably also have relevant thoughts/input here

jasmainak, 2022-01-19:
FEM seems like a complex undertaking, particularly because MNE is built around BEM-based forward modeling. It's something I have on the books after submitting the grant as it's proposed in it as well. I'd be happy to collaborate with you on that @alexrockhill . I haven't looked at particular software but some experimentation is needed to arrive at that

agramfort, 2022-01-20:
I've done BEM/FEM in the past with custom C++ code. I've never used these
libs but I will be happy to review/help
with what I know.

As it's learning while walking maybe you could start by working on a
separate repo before considering
contributing this to MNE

my 2c

> Message ID: ***@***.***>
>

alexrockhill, 2022-01-22:
Ok I was reading this PhD thesis which was helpful for describing the issues with software implementation https://research-repository.uwa.edu.au/files/94033696/THESIS_DOCTOR_OF_PHILOSOPHY_ZWICK_Benjamin_Fritz_2020.pdf

There is also a GitLab project written in Julia for their method but it doesn't contain an example for the brain deformation but hopefully wouldn't be too hard to extend. https://gitlab.com/benzwick/softisim

It sounds like he chose Julia because he didn't want to use Cython but needed the speed of C++. Is that something we'd be okay having in MNE (Cython that is)? That is if it ever ended up joining the main repo.

alexrockhill, 2022-01-22:
> I've done BEM/FEM in the past with custom C++ code. I've never used these libs but I will be happy to review/help with what I know. As it's learning while walking maybe you could start by working on a separate repo before considering contributing this to MNE my 2c

Do you still have the code around somewhere? That might be good to look at.

larsoner, 2022-01-22:
One option would be to see if you can get away with speeding up slow parts with Numba. But this requires reimplementation, so might not be worth it.

I don't think we want to add a C++ dependency to MNE-Python. We could think about doing it in another new sibling package like MNE-<something> that we can auto deploy to PyPi using CIBUILDWHEEL for example.

larsoner, 2022-01-22:
MNE-FEM would be a good name if it's really that general.

Also have you looked to see how Brainstorm solves these problems?

alexrockhill, 2022-01-22:
> Also have you looked to see how Brainstorm solves these problems?

BrainStorm doesn't do fem for contributions to the recording channel and projects to the surface of the brain for accounting for brain shift like we do now.

Let me dig into the package a bit and see how hard reimplementation would be. In the dissertation he mentioned he was learning to code but had a pretty advanced knowledge of computing so idk if he was just being modest and it is a huge undertaking to reimplement or if maybe the core algorithm is just a couple typical length functions.

alexrockhill, 2022-01-24:
Adding another resource that uses fenics but for volume conduction modeling https://github.com/meronvermaas/FEMfuns. I think it will be really helpful to see other implementations.

alexrockhill, 2022-01-27:
Does anyone have a good reference for a publication where finite element analysis was used to determine the tissue contributing to the recording? I would guess animal work with LFP and unit recordings might be particularly informative. @jasmainak? Maybe I'll look into more of Stephanie Jones' work...

jasmainak, 2022-01-27:
humm I don't know. Perhaps we should do a small video chat first to understand what you're trying to achieve. We could set up a time in March after the current grant cycle :)

alexrockhill, 2022-01-30:
> humm I don't know. Perhaps we should do a small video chat first to understand what you're trying to achieve. We could set up a time in March after the current grant cycle :)

Sounds great! I'd definitely be up for that

alexrockhill, 2022-03-22:
@jasmainak, do you still want to meet and talk about this?

sarangnemo, 2022-05-11:
You may want to check out the new FEM package [DUNEero](https://www.medizin.uni-muenster.de/duneuro/startseite.html), from the group that previously developed SimBio. They've made Python bindings, and it seems their Matlab bindings have already been integrated into FieldTrip and BrainStorm.

larsoner, 2022-06-19:
FYI some progress is being made toward #1624 which might help with sEEG forward computation

larsoner, 2026-03-04:
I noticed they have https://simnibs.github.io/simnibs/build/html/tutorial/advanced/eeg_leadfields.html#mne-python already! So maybe we "just" need an example that shows how to do some of this stuff?

*EDIT: Would be awesome if someone could make an example that does this for the `sample` dataset for example...*

sarangnemo, 2026-03-04:
> I noticed they have https://simnibs.github.io/simnibs/build/html/tutorial/advanced/eeg_leadfields.html#mne-python already! So maybe we "just" need an example that shows how to do some of this stuff?

I believe @jduemose was the one on the SimNIBS team working on this. (cc @axelthi and @ualsbombe)