mne-tools/mne-python #10221: Allow blocking execution of plot_alignment() and stc.plot()

hoechenberger, 2022-01-19:
The alignment and the `Brain` plots are both used in interactive exploration and in non-interactive scripts.

A common annoying issue that I and other users regularly run into is that we write a short script that displays the sensor alignment or a `SourceEstimate`, run it (e.g. via the `Run` button in VS Code), the window pops up – and immediately disappears. So one then either needs to run the script again -- this time, with `python -i` -- or inserts some blocking command like `input()`.

It would be great if there was an easier way to ensure a blocking behavior.

cc @GuillaumeFavelier

GuillaumeFavelier, 2022-01-19:
In the coreg app, we have a `standalone` parameter which control that. We could add the same to `Brain`? But adding yet another one to `plot_alignment` might be too much.

hoechenberger, 2022-01-19:
I was more thinking of something like `block`, which we have for our matplotlib plotting functions too

GuillaumeFavelier, 2022-01-19:
I do not mind changing the name if it does the same thing.

hoechenberger, 2022-01-19:
`standalone` is fine for Coreg, I think. Just for the others – if we add this functionality, I think it should be `block`. :)

GuillaumeFavelier, 2022-01-19:
Why keeping coreg as the only exception?

GuillaumeFavelier, 2022-01-19:
backward-compat... Honestly, the new version of coreg has not been around for so long, I don't know if people had really the time to play around with the `standalone` parameter. I still think the change is worth it for consitently (just a rename in practice). But I understand your point.

hoechenberger, 2022-01-19:
Ah ok! Thought we'd always had this param. In that case – let's **block** it 😁👌

GuillaumeFavelier, 2022-01-20:
#10222 is merged but the case of `plot_alignment` is still open.