mne-tools/mne-python #10227: Retrieving Coregistration from CoregistrationUI

hoechenberger, 2022-01-20:
#### Describe the new feature or enhancement
I need to programmatically retrieve the coregistration results from `CoregistrationUI` after the user has finished coregistration, without requiring them to explicitly save `trans` and `fiducials` to disk.

Currently, there's no public API to do so: the `Coregistration` instance is stored in `CoregistrationUI._coreg`.

#### Describe your proposed implementation
`CoregistrationUI._coreg` should become public as `CoregistrationUI.coreg`.

#### Describe possible alternatives
`mne.gui.coregistration()` could return the `Coregistration` instance after `CoregistrationUI` has been closed, instead of returning `CoregistrationUI`. But I believe this would break some workflows.

cc @GuillaumeFavelier @larsoner

agramfort, 2022-01-20:
I would add a public method to CoregistrationUI

> Message ID: ***@***.***>
>

hoechenberger, 2022-01-20:
> I would add a public method to CoregistrationUI

This could be another option, yes. Something that returns a tuple: a dict with fiducials (name->coords), and the head->mri Transform

GuillaumeFavelier, 2022-01-20:
Why not using `Coregistration` in the first place? It is public and we added it for this purpose in the first place (I mean scripting coregistration).

My work on coregistration started in https://github.com/mne-tools/mne-python/pull/9516 because `jhouck` wanted to [programmatically use the coreg UI](https://github.com/mne-tools/mne-python/pull/6693/files).

I would prefer that we improve/refactor `Coregistration` if it lacks features. And on the topic of trans, it actually has a [`trans` property](https://github.com/mne-tools/mne-python/blob/main/mne/coreg.py#L1951-L1954) which does not save to disk.

hoechenberger, 2022-01-20:
> Why not using `Coregistration` in the first place?

My use case is that we perform an interactive coregistration via `mne.gui.coregistration()`, and after that's done, I want to write the results (transformation and fiducials) to BIDS. To make this work reliably, I'd like to be able to directly extract this information from the `CoregistrationUI` object somehow, without forcing the user to create files first – whose filenames I cannot control.

Does that make sense?

GuillaumeFavelier, 2022-01-20:
> Does that make sense?

If `CoregistrationUI` must be used then indeed, at the moment, there is no public way to do that, I understand :ok_hand:

hoechenberger, 2022-01-20:
I can give this a shot if you don't mind

GuillaumeFavelier, 2022-01-20:
You're more than welcome to do so @hoechenberger

PR #10255 (2022-01-26): Publicly expose Coregistration instance from CoregistrationUI
This exposes `Coregistration` via `CoregistrationUI.coreg`

Closes #10227
commit abf530f7 (2022-01-26): Publicly expose Coregistration instance from CoregistrationUI (#10255)

* Publicly expose Coregistration instance from CoregistrationUI

This exposes Coregistration via CoregistrationUI.coreg

Fixes #10227

* Update mne/gui/_coreg.py

* Style & doc fixes
closed 2022-01-26 by larsoner