mne-tools/mne-python #10232: default verbose level in raw.filter() should not be None

SophieHerbst, 2022-01-20:
Version: 1.0.dev0

In raw.filter() the default for the verbose parameter is set to None, i.e. no comments are printed. 
This used to be different, and I think it is useful to provide basic information about the filter, for instance for reporting in a paper.

larsoner, 2022-01-20:
> This used to be different, and I think it is useful to provide basic information about the filter, for instance for reporting in a paper.

I think it has been `None` for a while. Is it possible that you on your system set `set_config('MNE_LOGGING_LEVEL', 'warning')` at some point?

In any case, I'm -1 on changing the default here -- it makes our software less consistent in how `verbose` and logging levels work for the user.

SophieHerbst, 2022-01-20:
> I think it has been None for a while. Is it possible that you on your system set set_config('MNE_LOGGING_LEVEL', 'warning') at some point?

Not intentionally, unless it is somehow set during installation via the environment file that was provided for previous versions?

> In any case, I'm -1 on changing the default here -- it makes our software less consistent in how verbose and logging levels work for the user.

I am not aware of your general approach to those defaults, but right now, per default, the filter function prints no output at all, which appears inconsistent with other functions, for instance maxwell_filter. Personally, I find it very useful to have information like transition bandwidths printed by default so the user understands that those values are set under the hood.

larsoner, 2022-01-20:
> the filter function prints no output at all, which appears inconsistent with other functions, for instance maxwell_filter. 

The other possibility is that you did `read_raw_fif(..., verbose=False)` or so. The `verbose` becomes part of `raw` and then you need to override it in the call to `filter`. This is another behavior we've had around for a while, but perhaps isn't documented clearly enough?

SophieHerbst, 2022-01-20:
Indeed, I used verbose='error' during import, not knowing that this gets attached to the raw further on. Thank you for pointing this out!
Indeed, it would help to document this more explicitly, for instance in the description of the verbose parameter in io.read_raw_fif.

SophieHerbst, 2022-01-21:
Hi @larsoner, I do not know what you mean by 'our docdict'. Can you point me to it?

hoechenberger, 2022-01-21:
> Indeed, I used verbose='error' during import, not knowing that this gets attached to the raw further on. Thank you for pointing this out!

🤯 I wasn't aware of this either!!! I commonly use `verbose='error'` when reading files because otherwise, I drown in output!

hoechenberger, 2022-01-21:
> Hi @larsoner, I do not know what you mean by 'our docdict'. Can you point me to it?

In `mne/utils/docs.py` we keep a huge dictionary full of (partial) docstrings that are shared between two or more functions, in an attempt to avoid duplication. We call this thing "the docdict", probably because that's the name of the variable: `docdict` 😅

mmagnuski, 2022-01-25:
> > Indeed, I used verbose='error' during import, not knowing that this gets attached to the raw further on. Thank you for pointing this out!
> 
> 🤯 I wasn't aware of this either!!! I commonly use `verbose='error'` when reading files because otherwise, I drown in output!

Haha, me too, I didn't know about this raw-verbose communion, but it sounds cool!

larsoner, 2023-01-02:
We no longer have `raw.verbose` affect method verbosity so I think we can close this!

closed 2023-01-02 by larsoner