mne-tools/mne-python #10254: Add anchors and / or tick labels to our QSlider and QFloatSlider instances

hoechenberger, 2022-01-26:
It's important to label at least the extrema (anchors) of range sliders for a better UX

It's not difficult to do:
https://stackoverflow.com/questions/27661877/qt-slider-widget-with-tick-text-labels

cc @GuillaumeFavelier 

Not urgent; in fact, I might just try to get this to work this weekend or so. So don't feel pressured to work on this.

larsoner, 2022-01-26:
Just keep in mind anything you do for Qt you should try to do for the notebook backend, too

Dev-satish, 2023-02-15:
@larsoner @hoechenberger can you please provide direction for this issue as it seems very easy to approach for a first time contributor in MNE

larsoner, 2023-02-15:
For this you'd want to look into `mne/viz/_brain` code for Qt and Notebook. But to understand the underlying problem, running through some examples of plotting source time courses and seeing the UI yourself is critical

Dev-satish, 2023-02-15:
Okay! I'll give it a shot incase I encounter any issue I'll reach out here again.

Dev-satish, 2023-02-18:
@larsoner how do I view the UI? I am able to run the tests successfully but on what command can I access the UI of this project? My apologies if this is a bit too rookie question.

drammock, 2023-02-18:
@Dev-satish there isn't one single UI. But the best place to start is probably the one for `brain.plot()`. Search the docs for tutorials that use that one