mne-tools/mne-python #10256: raw object meas_date is moderated by time zone of the computer reading the EEG file (from an EGI MFF file).

scott-huberty, 2022-01-26:
#### Describe the bug
the `Info['meas_date']` `datetime` object derived from a EGI Net station MFF file (continuous recording), appears to be moderated by the time zone of the computer which read in the EGI file. `info['meas_date']` should only be concerned with the timestamp and timezone indicated in the MFF file, and should be completely agnostic to the timezone of the computer reading in the MFF file.

#### Steps to reproduce
- I have imported the same EGI MFF file using the same code (`mne.io.read_raw_egi()`) on two separate computers, which we'll call **computer 1** and **computer 2**.
-See the two attached PDF's, for the code and output for each of the two computers
- The computers are on the same time zone (EST), but **Computer 1** is expressed in  Universal time (UTC) and **computer 2** is expressed in local time (EST).
- FYI that the MFF file was recorded in San Diego (PST).
- On **computer 1**, raw.info['meas_date'] returns `datetime.datetime(2019, 10, 29, 12, 41, 1, tzinfo=datetime.timezone.utc)`, indicating that the recording start time was **12:41pm**.
- On **computer 2**, `raw.info['meas_date']` returns `datetime.datetime(2019, 10, 29, 16, 41, 1, tzinfo=datetime.timezone.utc)`, indicating that the recording start time was **4:41pm**.
- The `datetime` returned by **computer 1** is the correct recording start time, so it would seem **computer 2** being set to local time caused the issue. Regardless, `info['meas_date']` should be completely agnostic to the timezone of the computer reading in the MFF file. 
- As far as I can tell, the same version of mne is used on both computers.
- I can share the MFF file used  for this report, upon request.

#### Expected results
`datetime.datetime(2019, 10, 29, 12, 41, 1, tzinfo=datetime.timezone.utc)`

Which was returned by **Computer 1**, and  I have confirmed that this is the correct start time of the recording (12:41pm)

[image]

#### Actual results
`datetime.datetime(2019, 10, 29, 16, 41, 1, tzinfo=datetime.timezone.utc)`

Which was returned by **Computer 2**, indicating that the recording start time was at 4:41pm, which is incorrect.
[image]

#### Additional information
**Computer 1**
Platform:       Linux-3.10.0-1160.36.2.el7.x86_64-x86_64-with-centos-7.9.2009-Core
Python:         3.7.4 (default, Jul 18 2019, 19:34:02)  [GCC 5.4.0]
Executable:     /cvmfs/soft.computecanada.ca/easybuild/software/2017/Core/python/3.7.4/bin/python
CPU:            : 24 cores
Memory:         Unavailable (requires "psutil" package)
mne:            0.24.1
numpy:          1.20.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:          1.4.1
matplotlib:     3.2.1 {backend=module://ipykernel.pylab.backend_inline}

**Computer 2**
Platform:       Linux-4.18.0-305.25.1.el8_4.x86_64-x86_64-with-glibc2.2.5
Python:         3.8.10 (default, Jun 16 2021, 14:20:20)  [GCC 9.3.0]
Executable:     /home/shuberty/virtual_envs/pylossless/bin/python
CPU:            : 64 cores
Memory:         Unavailable (requires "psutil" package)
mne:            0.24.1
numpy:          1.21.2 {blas=/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/flexiblas/3.0.4/include/flexiblas, lapack=/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/flexiblas/3.0.4/include/flexiblas}
scipy:          1.7.3
matplotlib:     3.4.2 {backend=module://matplotlib_inline.backend_inline}

[computer1_bugreport.pdf](https://github.com/mne-tools/mne-python/files/7943326/helios_bugreport.pdf)
[computer2_bugreport.pdf](https://github.com/mne-tools/mne-python/files/7943329/narval_bugreport.pdf)

Cheers,

Scott

larsoner, 2022-01-26:
This does seem like a bug -- `read_raw_egi` should detect any time zone offset, remove it to get to UTC, and save the actually used time zone in `info['utc_offset']`. Do you want to try fixing this? I think it would need to be around here:

https://github.com/mne-tools/mne-python/blob/b5688906f971ad0b4026d9812e021dbfd015bf70/mne/io/egi/egimff.py#L221-L222

and 

https://github.com/mne-tools/mne-python/blob/b5688906f971ad0b4026d9812e021dbfd015bf70/mne/io/egi/egimff.py#L456-L460

scott-huberty, 2022-01-26:
Sure, I can take a look at this in the next week or so, and will follow up.

Thanks for the help,

Scott