mne-tools/mne-python #10264: DOC: Description of picks in `ica.plot_properties` is incorrect

eort, 2022-01-28:
With the `picks` keyword in `ica.plot_properties`, one can specify what ica sources to plot the properties of. So picks should code the ica source. However, at the moment, the description of picks is the [standard phrase](https://mne.tools/stable/generated/mne.preprocessing.ICA.html#mne.preprocessing.ICA.plot_properties): 

> Channels to include. Slices and lists of integers will be interpreted as channel indices. In lists, channel type strings (e.g., ['meg', 'eeg']) will pick channels of those types, channel name strings (e.g., ['MEG0111', 'MEG2623'] will pick the given channels. Can also be the string values “all” to pick all channels, or “data” to pick data channels. 

plus something specific to this plot: 

> None (default) will pick the first five sources. If more than one components were chosen in the picks, each one will be plotted in a separate figure.

#### Proposed documentation ehancement
The first part does not make sense here, and should be replaced by something like: 

> Sources to plot. None (default) will pick the first five sources. If more than one components were chosen in the picks, each one will be plotted in a separate figure.

hoechenberger, 2022-01-28:
Good catch, wanna give it a shot?

eort, 2022-01-29:
Sure, will do!

PR #10273 (2022-01-31): DOC: change picks description in ICA.plot_properties()
Fixes #10264.

Like discussed, I updated the description what info picks represent.
closed 2022-01-31 by larsoner