mne-tools/mne-python #10285: ENH: Montage plots for fNIRS

larsoner, 2022-02-01:
Not sure if this should go in MNE-Python, MNE-NIRS, or neither (?), but I've repeatedly wanted to show (in papers, and just to make sure I'm doing things correctly) plots of fNIRS electrodes labeled with the 10-20 coordinates they have when they are being positioned on the subject. However:

1. fNIRS channels have specific namings related to their source-detector pairings and wavelengths, so `rename_channels` is not a good option
2. The channels we actually record are a combination of a source-detector pair, like the yellow dots [here](https://mne.tools/dev/auto_tutorials/preprocessing/70_fnirs_processing.html#viewing-location-of-sensors-over-brain-surface):

    [image]

3. However, the optodes that are placed on the heads are actually sources and detectors, i.e., the red and dark red/brown above.

What I've found really useful for sanity checking is a multi-view plot with ideal 10-20 locations, plus the resulting fNIRS channel numbers (starting from 1; here for a Hitachi system):

[image]

For MNE-Python, it might be useful to add text labels to channels in `plot_alignment`, though I'm not sure how nicely these could be made to interact with 3D view manipulations. (My code currently uses the given 3D-to-2D projection to add 2D text actors. Proper 3D interaction would be tougher, since we'd need to place the text in the 3D scene -- probably with some offset "outward" so that it doesn't overlap with the channel sphere -- and make sure it draws nicely with translucent geometry and such).

So I think overall I'm inclined to just implement some sort of `mne_nirs.viz.plot_3d_montage` or something. If folks agree about that scope, then I'll transfer this issue over to the MNE-NIRS repo and hash out an API with @rob-luke.

agramfort, 2022-02-01:
I have a mixed feeling on this. Is it a lot of code?

As I would expect users to be willing to do raw.plot_montage() like we do
for EEG and MEG

> Message ID: ***@***.***>
>

larsoner, 2022-02-01:
> I have a mixed feeling on this. Is it a lot of code?

About 60 lines, using `mne.viz.Brain` and pyvista/VTK methods.

> As I would expect users to be willing to do raw.plot_montage() like we do for EEG and MEG

This is of course possible, but:

1. It does not capture the source-detector-1020 relationship at all, just the "channel" locations (midpoint of source-detector) are shown
2. It does not show the relationship to the brain, which `plot_alignment` does nicely

The `raw.plot_sensors(show_names=True)` is less useful / clear about what was where and what fNIRS channels it gave rise to:

| topomap | 3d |
|-|-|
| [image] | [image] |

One way forward is to implement in MNE-NIRS, then if people think parts would be useful for M/EEG more generally, port to MNE-Python?