mne-tools/mne-python #10286: confusing warning with applying projs

jasmainak, 2022-02-01:
#### Describe the bug
Using `apply_proj` after marking bad channels should automatically do the right thing instead of raising a confusing warning.

#### Steps to reproduce

```py
import os
import numpy as np
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw = raw.pick_types(meg='mag', eeg=False)

print(raw.info['projs'])

raw.info['bads'] = raw.info['ch_names'][:16]
raw2 = raw.copy()

raw.apply_proj()
raw2.info.normalize_proj()
raw2.apply_proj()
np.testing.assert_allclose(raw.get_data(), raw2.get_data())
```

As per the warning, `normalize_proj` should be applied but as you can see, it makes no difference except getting rid of the warning.

#### Expected results
There should be no warning

#### Actual results
We see a warning:

```py
/Users/mainak/Desktop/mne_proj_issue.py:16: RuntimeWarning: Projection vector "PCA-v1" has magnitude 0.99 (should be unity), applying projector with 86/102 of the original channels available may be dangerous, consider recomputing and adding projection vectors for channels that are eventually used. If this is intentional, consider using info.normalize_proj()
  # raw.apply_proj()
/Users/mainak/Desktop/mne_proj_issue.py:16: RuntimeWarning: Projection vector "PCA-v2" has magnitude 0.95 (should be unity), applying projector with 86/102 of the original channels available may be dangerous, consider recomputing and adding projection vectors for channels that are eventually used. If this is intentional, consider using info.normalize_proj()
  # raw.apply_proj()
/Users/mainak/Desktop/mne_proj_issue.py:16: RuntimeWarning: Projection vector "PCA-v3" has magnitude 0.85 (should be unity), applying projector with 86/102 of the original channels available may be dangerous, consider recomputing and adding projection vectors for channels that are eventually used. If this is intentional, consider using info.normalize_proj()
```

#### Additional information
Additionally, the code path is somewhat confusing to follow. There is an `if inplace` clause:

https://github.com/mne-tools/mne-python/blob/93a3be5d21028331a88c42ef9dfa10566bcd6d1f/mne/io/proj.py#L618-L625

which makes the function return without performing an SVD. I see that `_make_projector` is never called with `inplace=False`. Is this dead code? We discussed this with @mshamalainen and the right thing to do is:

- combine the projectors into a matrix with bad channels zeroed out
- compute SVD of this matrix and use it to make the new projector

There is nothing "dangerous" per se in doing this as far as we can see. It only makes the projection operator mathematically correct. That is: [P.P = P](https://en.wikipedia.org/wiki/Projection_(linear_algebra)#Idempotence)

larsoner, 2022-02-01:
> it makes no difference except getting rid of the warning... There is nothing "dangerous" per se in doing this as far as we can see. It only makes the projection operator mathematically correct. That is: P.P = P

I think the warning is useful. If you're curious, look into the `blame` and see why it was put there. But the short version IIRC is that, if a channel is *really* bad, it might (easily) completely dominate the projector calculation. For example, if you `compute_proj_raw`, the first component could be totally dominated by a bad channel's oscillations (and whatever happens to correlate with them). So the "first projector" you get when marking this channel bad will differ quite a bit from the "first projector" you get without marking it as bad.

larsoner, 2022-02-01:
Some quick links I found:

https://github.com/mne-tools/mne-python/issues/2760
https://github.com/mne-tools/mne-python/pull/2838

But there are probably other discussions, too

jasmainak, 2022-02-02:
I did look at #2838 cursorily before posting the issue. Thanks for the links.

> So the "first projector" you get when marking this channel bad will differ quite a bit from the "first projector" you get without marking it as bad.

True, but the warning is complaining if the user does the *right* thing: that is mark the bad channels to fix the "first projector". If the user does not mark the bad channels, those projectors will transfer the artifacts from bad channels to good channels and the warning is not raised. The warning should probably be raised when computing the projector if at all.

Also, `normalize_proj` does not do anything other than suppress the warning because `apply_proj` already does the right thing.

jasmainak, 2022-02-02:
Also:

```py
applying projector with 86/102 of the original channels available may be dangerous
```

seems to suggest that subselecting channels in projection operator is dangerous. Rather, what's dangerous is if there was a bad channel when computing the original projector which is a different problem

larsoner, 2022-02-02:
> True, but the warning is complaining if the user does the right thing: that is mark the bad channels to fix the "first projector".

My point is that in some (many? most?) cases, this is not the right thing to do. The right thing to do is to mark them bad before computing the projector at all, because the bad channel's presence can influence the PCA computed during SSP.

To me we are correct to warn, but we should improve the message

jasmainak, 2022-02-02:
Let's say you read a fif file that has projectors already added from the daily empty room. It will not have the bad channel marked most likely and you will have to mark it manually in MNE-Python. This will lead to the cryptic warning. What situations do you see that it's beneficial to warn when applying the projector rather than when computing the "original projector"?

larsoner, 2022-02-02:
> What situations do you see that it's beneficial to warn when applying the projector rather than when computing the "original projector"?

None, but there is no way that I can think of to warn someone *when they are initially computing SSP* that at some point in the future they might realize a channel is bad and mark it as such. We can just detect that they've marked a channel (or set of channels) bad after the fact in a way that reduces the power in the SSP (should have unit norm but no longer does, in some meaningful sense) when they go to apply it. Unless I'm missing what you're suggesting to do here?

larsoner, 2022-02-02:
> Let's say you read a fif file that has projectors already added from the daily empty room

Yes, in this case and some others it's not necessary to warn. In others out indicates a real problem with the SSP computation that the incorrect / mismatched set of bad channels was used.

larsoner, 2022-02-02:
(... and even in the case of a daily empty room projector, the warning is only spurious of the channel was actually good at the time when SSP was computed. If it was bad during SSP computation, too -- and large in amplitude -- it's possible even in that case that the first resulting projection vector is contaminated / wrong / less useful than it would be if the SSP had been calculated with the channel marked bad, because the first PCA component will be dominated by the one channel plus others to the extent that they correlated with it.)

larsoner, 2022-02-02:
... At least I think this is what would happen. I'll make a little synthetic example and we can see if it actually comes out that way.

agramfort, 2022-02-02:
2 remarks:
- it's possible that some system SSP have been computed before some
channels went bad. Is it likely?
- as soon as you applied SSPs it becomes much hard to spot bad channels
usually. You should do mark
channels as bad as early as possible

Message ID: ***@***.***>
>

jasmainak, 2022-02-02:
> but there is no way that I can think of to warn someone when they are initially computing SSP that at some point in the future they might realize a channel is bad and mark it as such.

You can warn based on the eigen values because what you [suggested above](https://github.com/mne-tools/mne-python/issues/10286#issuecomment-1027327558) was that a bad channel might dominate the eigenvalue decomposition. But yes, we'd have to develop some threshold for this

> If it was bad during SSP computation, too -- and large in amplitude -- it's possible even in that case that the first resulting projection vector is contaminated / wrong / less useful than it would [...]

Yes, but see what the warning instructs the user to do. It asks the user to apply `normalize_proj`. It took me a while to figure out what `normalize_proj` does. If you follow the codepath carefully, it fixes the projector vector in some way but applying the projector to the data after doing `normalize_proj` is no different than doing `apply_proj` directly. You could as well call this a parameter `suppress_warning` in `apply_proj`

jasmainak, 2022-02-02:
For now, my vote is to completely nuke the warning and do the right thing by subselecting the projector so that it removes the current selection of bad channels. There is almost no way to know if the user is doing the right thing after the fact and `normalize_proj` does nothing other than remove the warning.

jasmainak, 2022-02-02:
> should have unit norm but no longer does, in some meaningful sense

By the way, this will happen no matter what when you mark a bad channel. Even if the channel was not originally too bad, it will reduce the norm

jasmainak, 2022-02-02:
Maybe you could warn/info:

> Detected bad channels that were not marked during SSP computation. Consider recomputing the SSP with the channels marked as bad. Subselecting the SSP to remove bad channels ...

Or some such thing

larsoner, 2022-02-02:
> You can warn based on the eigen values because what you suggested above was that a bad channel might dominate the eigenvalue decomposition. But yes, we'd have to develop some threshold for this

I guess we could try but sometimes large interference patterns show up on a small handful of channels (way fewer than 16!)... so we'll have another threshold issue (as we do now, see below). So to me it's not all that different.

> it will reduce the norm

Yes by some amount. We currently have some threshold (1% or something I think?) that we could in principle adjust to be less sensitive. Maybe some percentage relative to the number of channels in the vector would work better.

> For now, my vote is to completely nuke the warning and do the right thing

You keep repeating "do the right thing". This is the part I disagree with. Sometimes it's the right thing, yes, but sometimes it's wrong, and the user should do something else or they will get an incorrect result. Do you agree with this / see my point? It sounds like you still don't.

But if you do agree that this can happen, then in situations where the user might have (easily, often enough) done something bad and we cannot really tell, we typically warn. False alarms are safer than misses in this sense. So I'd like to keep the warning.

> Maybe you could warn/info

Yes, I'm all for making the warning clearer! What we suggest is not at all clear that the user might be okay *or* might be doing something wrong, depending on the circumstances

agramfort, 2022-02-02:
@jasmainak you suggest here an alternative way to compute the projectors
right?

> Message ID: ***@***.***>
>

larsoner, 2022-02-02:
> you suggest here an alternative way to compute the projectors right?

I think that's this:

> combine the projectors into a matrix with bad channels zeroed out
>
> compute SVD of this matrix and use it to make the new projector

But I expect this to be equivalent to what we already do...?

larsoner, 2022-02-02:
> It is different because it raises the warning at the right point. When the user is at apply_proj they should not be told to go back and recompute the proj. 

Ideally, yes, but I suspect it will be harder to catch this early, rather than after the user does the risky thing (marking channels bad post-computation).

> I'm still trying to wrap my head around what exactly is wrong with the warning... But what is that "something else" that the user should do if it's the wrong thing?  ... The only thing they could fix is to go back and mark bad before computing the projector ... that's a totally different warning than what is being printed right now.

Yes exactly, the safest thing is that the user should mark the channels bad and recompute SSP. Here is an example about why it matters:

<details>

```
import numpy as np
import mne
from sklearn.decomposition import PCA
import matplotlib.pyplot as plt

# Create some uncorrelated data
n_ch, n_samp, sfreq = 10, 10000, 1000.
data = np.random.RandomState(0).randn(n_ch, n_samp)
data -= np.mean(data, axis=-1, keepdims=True)
data = np.linalg.svd(data, full_matrices=False)[2]
np.testing.assert_allclose(np.corrcoef(data), np.eye(n_ch), atol=1e-6)

# Add one correlated component to it that we want to suppress (noise)
ideal_proj = np.zeros(n_ch)
ideal_proj[:-1] = np.hanning(n_ch - 1)
ideal_proj /= np.linalg.norm(ideal_proj)
t = np.arange(n_samp) / sfreq
sig_noise = np.sin(2 * np.pi * 10 * t) * ideal_proj[:, np.newaxis]
data += sig_noise

info = mne.create_info(n_ch, sfreq, 'eeg')
raw = mne.io.RawArray(data, info)

def compute_plot_proj(raw, ax, title, bad=None):
    assert bad in (None, 'before', 'after')
    if bad == 'before':
        raw.info['bads'] = [raw.ch_names[-1]]
    else:
        raw.info['bads'] = []
    proj = mne.compute_proj_raw(raw, n_eeg=1)
    ax.plot(ideal_proj, color='k', lw=2, zorder=4)
    plot_data = proj[0]['data']['data'][0]
    if bad == 'after':
        # trim last channel and renormalize (what our code does)
        plot_data = plot_data[:-1]
        plot_data /= np.linalg.norm(plot_data)
    plot_data *= np.sign(np.mean(plot_data))  # flip sign to positive
    ax.plot(plot_data, color='r', lw=1, zorder=5)
    ax.set(title=title, xlabel='Channel', xlim=[-0.5, n_ch])
    ax.set_xticks(np.arange(n_ch))

# Clean data looks okay
fig, axes = plt.subplots(1, 4, figsize=(8, 2), constrained_layout=True)
compute_plot_proj(raw, axes[0], title='Clean data')

# Make the last channel very bad, but don't mark it: just captures last channel
raw._data[-1] += 10 * np.sin(2 * np.pi * 20 *t)
compute_plot_proj(raw, axes[1], title='Bad channel')

# Mark it bad after computation: still looks bad
compute_plot_proj(raw, axes[2], title='Marking after SSP', bad='after')

# Mark it bad before computation: looks good
compute_plot_proj(raw, axes[3], title='Marking before SSP', bad='before')
```

</details>

Black is the actual interference pattern (by construction) / ideal projector weight (as a function of channel index), red is the first computed projector under different scenarios:

[image]

(FWIW, based on how I added the bad channel, the projector in the "mark after" case will probably just reflect the 20 Hz power in each channel!)

Hence to me the quick, easy path forward for now is just to update the warning with something that makes it clear that, as @agramfort says, "you should mark bad channels as early as possible", and in many cases, should recompute SSP.

larsoner, 2022-02-02:
> I see that _make_projector is never called with inplace=False. Is this dead code?

One last point to clarify here -- note that the default is `inplace=False`:

https://github.com/mne-tools/mne-python/blob/258b462f06a3a5ef4c6d8ded6ca03f519b498831/mne/io/proj.py#L542-L543

So for example this bit of code in `mne.io.proj.make_projector` should hit the `inplace=False` path:

https://github.com/mne-tools/mne-python/blob/258b462f06a3a5ef4c6d8ded6ca03f519b498831/mne/io/proj.py#L539

jasmainak, 2022-02-02:
Thanks, yes I understood the `inplace` codepath after posting the issue but was missing the normalizing bit which missed my attention

PR #10292 (2022-02-12): ENH: Better logic and message
A concrete proposal for #10286:

1. Make the check depend on a 10%+ reduction in projector norm, which is better than looking for a 10% channel count reduction, and 
2. Improve the warning to make it clear that this can be problematic, or not

An alternative idea of adding a `suppress_proj_warning` or so has also been suggested, but I'm -1 on that because I think there are probably dozens of places we'd need to add it (i.e., anywhere that we apply proj -- creating epochs, creating evoked, plotting functions, cov computation, inv computation, ...).

Another alterna [... cut]
closed 2022-02-12 by agramfort