mne-tools/mne-python #10287: set_eeg_reference with projection=True and make_eeg_average_ref_proj only works with EEG channels

chapochn, 2022-02-01:
#### Describe the bug
when channels in raw are not eeg, but for example ecog then:
```
raw.set_eeg_reference(ref_channels='average', projection=False,
                                 ch_type='auto')
```
works, but:
```
raw.set_eeg_reference(ref_channels='average', projection=True,
                                 ch_type='auto')
```
does not, see full error below.

#### Steps to reproduce
Example
```
info = mne.create_info(ch_names=['a', 'b'], sfreq=100)
raw = mne.io.RawArray(np.zeros((2, 10)), info)
raw.set_channel_types({'a':'ecog', 'b':'ecog'})
# raw.set_channel_types({'a':'eeg', 'b':'eeg'}) # this works
raw.set_eeg_reference(ref_channels='average', projection=True, ch_type='auto')
raw.set_eeg_reference(ref_channels='average', projection=False, ch_type='auto') # this works with ecog too
```

#### Expected results
that a new channel that is the average projection is created

#### Actual results
fails with the following error:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/var/folders/w9/3c0pg_qj35x4m8cq745x1txxj4dh2s/T/ipykernel_92274/3378246075.py in <module>
      3 raw = mne.io.RawArray(np.zeros((2, 10)), info)
      4 raw.set_channel_types({'a':'ecog', 'b':'ecog'})
----> 5 raw.set_eeg_reference(ref_channels='average', projection=True,
      6                                  ch_type='auto')

<decorator-gen-34> in set_eeg_reference(self, ref_channels, projection, ch_type, forward, verbose)

~/opt/anaconda3/envs/mne/lib/python3.9/site-packages/mne/channels/channels.py in set_eeg_reference(self, ref_channels, projection, ch_type, forward, verbose)
    338         """
    339         from ..io.reference import set_eeg_reference
--> 340         return set_eeg_reference(self, ref_channels=ref_channels, copy=False,
    341                                  projection=projection, ch_type=ch_type,
    342                                  forward=forward)[0]

<decorator-gen-272> in set_eeg_reference(inst, ref_channels, copy, projection, ch_type, forward, verbose)

~/opt/anaconda3/envs/mne/lib/python3.9/site-packages/mne/io/reference.py in set_eeg_reference(inst, ref_channels, copy, projection, ch_type, forward, verbose)
    324                     inst.info['custom_ref_applied'] = \
    325                         FIFF.FIFFV_MNE_CUSTOM_REF_OFF
--> 326                 inst.add_proj(make_eeg_average_ref_proj(inst.info,
    327                                                         activate=False))
    328             except Exception:

<decorator-gen-14> in make_eeg_average_ref_proj(info, activate, verbose)

~/opt/anaconda3/envs/mne/lib/python3.9/site-packages/mne/io/proj.py in make_eeg_average_ref_proj(info, activate, verbose)
    772     n_eeg = len(eeg_sel)
    773     if n_eeg == 0:
--> 774         raise ValueError('Cannot create EEG average reference projector '
    775                          '(no EEG data found)')
    776     vec = np.ones((1, n_eeg))

ValueError: Cannot create EEG average reference projector (no EEG data found)
```

#### Additional information
Platform:       macOS-11.6-x86_64-i386-64bit
Python:         3.9.7 | packaged by conda-forge | (default, Sep 29 2021, 20:33:18)  [Clang 11.1.0 ]
Executable:     /Users/.../opt/anaconda3/envs/mne/bin/python3
CPU:            i386: 12 cores
Memory:         16.0 GB

mne:            0.24.1
numpy:          1.21.4 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:          1.7.3
matplotlib:     3.5.1 {backend=module://matplotlib_inline.backend_inline}

sklearn:        1.0.1
numba:          0.53.1
nibabel:        3.2.1
nilearn:        0.8.1
dipy:           1.4.1
cupy:           Not found
pandas:         1.3.5
mayavi:         4.7.2
pyvista:        0.32.1 {OpenGL 4.1 ATI-4.6.20 via AMD Radeon Pro 560X OpenGL Engine}
pyvistaqt:      0.6.0
ipyvtklink:     Not found
vtk:            9.0.3
PyQt5:          5.12.3
ipympl:         Not found
mne_qt_browser: 0.1.7
pooch:          v1.5.2

agramfort, 2022-02-02:
yes this is currently not supported. You absolutely need to do it with a
projection?

> Message ID: ***@***.***>
>

chapochn, 2022-02-02:
It feels useful to have that projection saved as a channel for future use and be able to analyze/plots things with and without the reference subtracted. I felt that it seems liked an unnecessary restriction... Is it a lot of work to just remove that restriction?

larsoner, 2022-02-02:
> Is it a lot of work to just remove that restriction?

I think in most places we have historically restricted things to EEG we have more recently relaxed / enhanced things to work with ECoG/sEEG/DBS as well. This seems like a logical place for it, too.

I don't foresee it causing too many problems. Do you want to take a stab at it @chapochn ?

chapochn, 2022-02-02:
Sure I'll check

larsoner, 2022-12-08:
I think this is available in 1.2

closed 2022-12-08 by larsoner