mne-tools/mne-python #10295: Footer on website requires scrolling down

cbrnr, 2022-02-03:
I'm not sure if this intentional, but no matter how tall my browser window is, it never shows the footer. Personally I find this not ideal because it means that a vertical scrollbar is always shown even though there would be enough available space.

agramfort, 2022-02-03:
I don't get what you mean

> Message ID: ***@***.***>
>

cbrnr, 2022-02-04:
I mean the line at the bottom that currently says "© Copyright 2012–2021, MNE Developers. Last updated Dec 3, 2021, 6:52:28 AM GMT+1". When you load the page, you don't see it. You always have to scroll down even when there is enough vertical space. In that case, I would show the footer instead of having users scroll down.

cbrnr, 2022-02-04:
https://user-images.githubusercontent.com/4377312/152548642-98c5025c-a6b4-4aea-8ce4-d692af11c581.mov

larsoner, 2022-02-04:
Closing as dup of https://github.com/pydata/pydata-sphinx-theme/issues/497

closed 2022-02-04 by larsoner