mne-tools/mne-python #10310: Inline links could be more visible in the documentation

mmagnuski, 2022-02-08:
Maybe my eyesight is not good, but I think the inline links in the docs could be a little more distinguishable:
[image]

when pasted here it actually looks better than on my screen because the dark font of main text is a little brighter in the example above (interpolation, I think).

cbrnr, 2022-02-08:
Good old underline maybe?

larsoner, 2022-02-08:
Is it the same as https://pydata-sphinx-theme.readthedocs.io/en/latest/ ? If so it would be good to open an issue there first, and address here with CSS overrides only if they reject the idea of making the links clearer somehow

mmagnuski, 2022-02-13:
@larsonerYes, it looks the same. Ok, I'll try that next week!

larsoner, 2023-04-21:
They have adjusted this so I'll close. If it's still a problem let's work upstream!

closed 2023-04-21 by larsoner