mne-tools/mne-python #10311: Zero padded channel labels don't seem to read correctly in the electrodes.tsv file

Aaronearlerichardson, 2022-02-08:
#### Describe the bug
I have a BIDS complaint database that I created to use mne's data processing pipelines on. The `read_raw_bids` function executes relatively successfully, but doesn't seem to keep all of my channel locations intact. Upon investigation, it appears that this is because some of the channel labels have zero padding to keep the names consistent (a practice often used in the BIDS specification), and that channels with said padding end up lost.

#### Steps to reproduce
Here is the contents of the electrodes.tsv file in question:
```
name	x	y	z	size	hemisphere
LAMT12	-49.0	4.333	-22.667	0.2	LD
LAMT11	-46.0555	4.333	-22.7225	0.2	LD
LAMT10	-43.111	4.333	-22.778	0.2	LD
LAMT09	-40.1665	4.333	-22.8335	0.2	LD
LAMT08	-37.222	4.333	-22.889	0.2	LD
LAMT07	-34.2775	4.333	-22.9445	0.2	LD
LAMT06	-31.333	4.333	-23.0	0.2	LD
LAMT05	-27.9522	4.3806	-23.1428	0.2	LD
LAMT04	-24.5714	4.4282	-23.2856	0.2	LD
LAMT03	-21.1906	4.4758	-23.4284	0.2	LD
LAMT02	-17.8098	4.5234	-23.5712	0.2	LD
LAMT01	-14.429	4.571	-23.714	0.2	LD
LAST08	-54.0	-2.667	-9.667	0.2	LD
LAST07	-50.333	-2.667	-10.0	0.2	LD
LAST06	-46.667	-2.667	-10.0	0.2	LD
LAST05	-43.4753	-2.7003	-9.975	0.2	LD
LAST04	-40.2835	-2.7335	-9.95	0.2	LD
LAST03	-37.0917	-2.7668	-9.925	0.2	LD
LAST02	-33.9	-2.8	-9.9	0.2	LD
LAST01	-30.55	-2.8222	-9.9388	0.2	LD
LPO10	-37.0	-80.333	16.333	0.2	LD
LPO09	-34.8	-78.4	14.4	0.2	LD
LPO08	-32.72	-76.4	12.48	0.2	LD
LPO07	-30.64	-74.4	10.56	0.2	LD
LPO06	-28.56	-72.4	8.64	0.2	LD
LPO05	-26.48	-70.4	6.72	0.2	LD
LPO04	-24.4	-68.4	4.8	0.2	LD
LPO03	-22.0	-66.2	2.8	0.2	LD
LPO02	-19.8571	-64.181	0.8667	0.2	LD
LPO01	-17.7143	-62.162	-1.0666	0.2	LD
LAO16	-54.2778	-68.7778	14.0	0.2	LD
LAO15	-50.8889	-67.8889	14.5	0.2	LD
LAO14	-47.5	-67.0	15.0	0.2	LD
LAO13	-44.0	-66.5	16.0	0.2	LD
LAO12	-41.0	-65.5	16.0	0.2	LD
LAO11	-37.5714	-64.5714	16.5	0.2	LD
LAO10	-34.1429	-63.6429	17.0	0.2	LD
LAO09	-30.7143	-62.7143	17.5	0.2	LD
LAO08	-27.2857	-61.7857	18.0	0.2	LD
LAO07	-23.8571	-60.8571	18.5	0.2	LD
LAO06	-20.4286	-59.9286	19.0	0.2	LD
LAO05	-17.0	-59.0	19.5	0.2	LD
LAO04	-13.6111	-58.1111	20.0	0.2	LD
LAO03	-10.2222	-57.2222	20.5	0.2	LD
LAO02	-6.8333	-56.3333	21.0	0.2	LD
LAO01	-3.4444	-55.4444	21.5	0.2	LD
LPIT16	-54.5417	-45.625	-4.5833	0.2	LD
LPIT15	-52.0	-45.5	-4.0	0.2	LD
LPIT14	-48.5	-46.0	-3.0	0.2	LD
LPIT13	-46.15	-45.85	-2.5	0.2	LD
LPIT12	-43.8	-45.7	-2.0	0.2	LD
LPIT11	-41.45	-45.55	-1.5	0.2	LD
LPIT10	-39.1	-45.4	-1.0	0.2	LD
LPIT09	-36.75	-45.25	-0.5	0.2	LD
LPIT08	-34.4	-45.1	0.0	0.2	LD
LPIT07	-32.05	-44.95	0.5	0.2	LD
LPIT06	-29.7	-44.8	1.0	0.2	LD
LPIT05	-27.35	-44.65	1.5	0.2	LD
LPIT04	-25.0	-44.5	2.0	0.2	LD
LPIT03	-21.5	-44.0	3.0	0.2	LD
LPIT02	-18.9583	-43.875	3.5833	0.2	LD
LPIT01	-16.4167	-43.75	4.1667	0.2	LD
LPSP16	-58.167	-39.833	43.0	0.2	LD
[... 106 lines cut ...]
CPU:            Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 8 cores
Memory:         63.9 GB

mne:            0.24.1
numpy:          1.22.1 {}
scipy:          1.8.0
matplotlib:     3.5.1 {backend=QtAgg}

sklearn:        1.0.2
numba:          Not found
nibabel:        3.2.1
nilearn:        0.8.1
dipy:           1.4.1
cupy:           Not found
pandas:         1.3.5
mayavi:         4.7.4
pyvista:        0.33.2 {OpenGL 4.5.0 - Build 27.20.100.8854 via Intel(R) HD Graphics 630}
pyvistaqt:      0.7.0
ipyvtklink:     0.2.2
vtk:            9.1.0
PyQt5:          5.15.6
ipympl:         0.8.7
mne_qt_browser: Not found
pooch:          v1.4.0
```

agramfort, 2022-02-09:
can you be more specific? can you make the data available to see?

Message ID: ***@***.***>
>

mmagnuski, 2022-02-09:
@agramfort I think the problem is that, for example in the channel positions (`electrodes.tsv`) a channel is named `LPSP05` but in the data file (or somewhere else, I don't know BIDS well) it is `LPSP5`. This leads to a warning that some channels do not have their positions specified and consequently these positions are not defined. I don't know if the mismatch between the labels is a BIDS problem or not.

agramfort, 2022-02-09:
hum well in this case I don't see how mne-bids can do the right thing
automatically....

> Message ID: ***@***.***>
>

mmagnuski, 2022-02-09:
@Aaronearlerichardson
If the channel labels are different in the electrode positions file than in the data file, they will not be read correctly. You should either change the channel labels in the electrode positions file (removing the zero prefixes) or change the labels in the data file (adding the zero prefixes).

Aaronearlerichardson, 2022-02-09:
Ahah! I checked the .edf file header and you're right the labels were different. False alarm! Thank you for your time.

closed 2022-02-09 by Aaronearlerichardson