mne-tools/mne-python #10314: raw.plot() does not produce specified background color.

fishbacp, 2022-02-09:
#### Describe the bug
*raw.plot resluts in a graph that does not use specified background color*

Example:

```raw.plot(bgcolor='y',color='r')```

#### Expected results
Background of plot should be yellow and traces colored red.

#### Actual results
Background is white and traces are red.

#### Additional information
Experimentation with other background color options does not change the outcome.

mmagnuski, 2022-02-09:
As stated on [MNE Discourse discussion](https://mne.discourse.group/t/unable-to-control-background-colors-for-raw-plot/4385/2) I can reproduce the problem. I can take a look at this in around 5 hours, unless someone looks first. :)

mmagnuski, 2022-02-10:
I looked at this and `mne.raw.plot_raw()` ignores `bgcolor` (does not pass it down to lower-level functions)
I tried adding it to the parameters that are passed around but it seemed to have no effect either when added as `bgcolor` or `facecolor`. I didn't have time to fully trace it down, as the mne raw plotting code is quite complex now (at least comparing to what we had some time ago), but will try to take a look again this evening.

PR #10317 (2022-02-11): FIX: set background color in raw.plot()
#### Reference issue
Example: Fixes #10314.

#### What does this implement/fix?
`bgcolor` argument was ignored by raw plotter.

```python
import os.path as op
import mne

pth = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample')
fname = 'sample_audvis_raw.fif'
raw = mne.io.read_raw(op.join(pth, fname))
raw.plot(bgcolor='red')
```
on main branch the `bgcolor` does not lead to any observable effects. Now it looks like this:
[image]

#### TODOs:
- [ ] add test
- [ ] whats new
- [ ] check if other plotters need this change?
closed 2022-02-11 by larsoner