mne-tools/mne-python #10320: ENH: Return Path in `data_path`?

larsoner, 2022-02-10:
I saw @hoechenberger's code [here](https://github.com/mne-tools/mne-qt-browser/pull/43/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R38-R40):
```
sample_dir = mne.datasets.sample.data_path()
raw_path = Path(sample_dir) / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
```
And thought it was a bit sad that he had to `Path(...)` the `data_path()` result.

Since 1.0 is coming out, I think one small breaking change we could make would be to return `Path` instead of `str` in `data_path`.

1. It actually matches the name of the function better
2. It should help people modernize their code a bit with things like `...data_path() / 'subject'`
3. `op.join` still works and returns `str`

The big problem I see is:

1. This fails:
    ```
    >>> Path('.') + '/foo'
    TypeError: unsupported operand type(s) for +: 'PosixPath' and 'str'
    ```
    The fix is pretty easy in this case, but it will break people's code. The fix is pretty easy, but it's still some pain...

We can probably get around this, though, with some class magic -- we could subclass `PosixPath` and `WindowsPath` with support for a `+` operator, where if the other arg is a `str` then we cast to `str` and return `op.join(path_str, other_str)` (after stripping `op.sep` from the `other_str`).

Thoughts?

An alternative would be to do something like `data_path(..., out='path')`, but I'm not sure that's much better than having people import from `pathlib` (though I guess it's shorter). One advantage of that approach is that we could deprecate `out='str'` -> `out='path'` to give people time to change.

cbrnr, 2022-02-10:
I'm in favor of this change. With our usual deprecation cycle, I think it's OK if people need to update their code. I don't think you can expect to run an old analysis with the latest MNE-Python – for this use case, people can use older versions of MNE-Python. It's not worth adding a lot of custom code just to keep the old `+` behavior around.

cbrnr, 2022-02-10:
> Even internally we use it a lot, and I know from reading other people's scripts (and even some of my own when I get lazy) that it is pretty prevalent.

We will update our code, and people can change their scripts. I'm talking about *new* scripts, I don't think we need to stay backwards compatible just so people can run their 5 year old scripts with the latest MNE version.

larsoner, 2022-02-10:
> This sounds good, but I'm not sure if adding a parameter makes things simpler.

I don't think it makes things simpler, either, but it does allow us to move from pure str to pure Path via deprecation. Without a `out`-type of parameter, we can't properly deprecate. And if we don't deprecate, I think we're going to break a lot of stuff for a lot of people. Maybe it's okay since it's a 1.0 release, but still seems a bit painful...

> I don't like this idea, let's not introduce a new class here. This will just add loads of confusion.

What's the loads of confusion / can you describe what a novice or advanced user will experience that will be problematic? For advanced users, we can mention in the docstring that we return an enhanced version of `Path` that supports `+` with str for backward compat. For novice users, they probably won't notice that `+` continues to work in scripts they already have, and can transition to the correct Path syntax of `/` whenever they get around to it.

hoechenberger, 2022-02-10:
> This sounds good, but I'm not sure if adding a parameter makes things simpler. A `pathlib.Path` is supposed to be a drop-in replacement for a `str` representing a path. If people would like to retain the old behavior (after a few versions of warnings) they can do `str()` on the output.

Fine with me too, however the idea with the new parameter stems from the assumption that we would like to retain the current behavior (returning a string) for a few versions while **already** allowing users to also get a `Path` instead.

I would also be okay with removing this parameter again once we switch the default, but urgh, that's just another deprcation 😅😅🙈

hoechenberger, 2022-02-10:
@larsoner 
> > This sounds good, but I'm not sure if adding a parameter makes things simpler.
> 
> I don't think it makes things simpler, either, but it does allow us to move from pure str to pure Path via deprecation. Without a `out`-type of parameter, we can't properly deprecate.

But what's the argument against introducing such a parameter? (Maybe I missed it in the discussion? 🤔)

> > I don't like this idea, let's not introduce a new class here. This will just add loads of confusion.
> 
> What's the loads of confusion / can you describe what a novice or advanced user will experience that will be problematic?

I don't like the idea of introducing yet another way to handle paths. We (MNE users) already have strings and `os.path`; we have MNE-BIDS's `BIDSPath` with the `.fpath` attribute, which returns a `pathlib.Path`; and now we're talking about not only supporting `pathlib.Path`, but a subclass that deviates from the original. It just feels very messy to me. This would allow for 3 ways to concat paths:

```python
os.path.join(a, b)
a / b
a + b
```
IDK, I'm not convinced…

cbrnr, 2022-02-10:
If the black magic eventually goes away this approach is pretty good!

hoechenberger, 2022-02-10:
@larsoner 
> but we could:
> 
> 1. Return my magic subclass of Path that supports `+`, but also
> 2. Have the `+` operation emit a DeprecationWarning that `+` support will go away in 1.1, so users should use `/` instead

Sounds good to me!

jasmainak, 2022-02-10:
I would like to vote in favor of a proper deprecation cycle. The threshold for breaking changes in MNE-ecosystem is a bit too low at the moment. Even if you talk about one release cycle for deprecation, that's around 6 months. It's not too uncommon to come back to a script after 6 months or even a year. 5 years is maybe pushing it, but I'd love to see a script that ran 2 years ago to run today without any errors :)

cbrnr, 2022-02-11:
I see your point, but why is it not possible for users and us to use an older version to run older code? This will always work no matter how many breaking changes we introduce. I agree we should not do that in every new release, but keeping ugly workarounds just so that the latest version runs old scripts is not worth the extra maintenance effort IMO.

jasmainak, 2022-02-11:
But how do I build upon old code if I do not use the latest API? Maybe I don't need the latest function from MNE but I need the latest function from numpy to do this additional analysis but that version of numpy is not compatible with the old version of MNE. Of course, I could export certain files, import it in the new environment and then run that additional analysis but is that optimal from a user-perspective?

what might be considered ugly from a coding point of view is actually beautiful from a UX point of view: the lack of unnecessary errors and warnings. The questions is: how does it affect the user experience besides *looking* ugly?

cbrnr, 2022-02-11:
You build upon old code by replacing deprecated with supported code.

Again, you can *always* run old scripts that worked with a given MNE version by using *exactly that version*. If you work on a new script (which includes taking an old script as a basis) then you will have to update certain things.

cbrnr, 2022-02-11:
Also, remember that we are talking about a major version bump here – this is exactly meant for breaking API, so we could change it without deprecation. Semantic versioning implies that version 1.x.x is in general not compatible with 0.x.x.

jasmainak, 2022-02-11:
In the past several months, *whenever* I run an MNE-related script ... I have had to spend first 15 minutes of my coding time fixing API related changes. Anyhow, I am just trying to add user perspective which is usually missing from the github discussions :)

cbrnr, 2022-02-11:
Yes sure, but why don't you use the MNE version you used when you created the script? Ideally, you should already have an environment which preserves exact versions. But I know, that's probably not what the average user does.

Anyhow, this discussion is drifting off a bit, so to come back to the original issue and proposed solution, I think we all agree that @larsoner's suggestion both uses proper deprecation and in two versions will be gone without any leftovers for us to keep maintaining - so a perfect solution to me!

mmagnuski, 2022-02-11:
I'm also +1 one for the subclassing witchcraft that raises deprecation warning and goes away in some time in future.

closed 2022-02-16 by larsoner

hoechenberger, 2022-02-16:
Yay!

PR #10348 (2022-02-16): API: Return Path in data_path
Closes #10320

WIP because I still need to change ~200 instances in examples and tutorials. Pushing now to get CIs happy to make sure the simple magic class isn't breaking anything. The one thing I "broke" on purpose was `35_dipole_orientations.py` where `+` is used but I didn't change it yet, just to make sure the DeprecationWarning shows up there.