mne-tools/mne-python #10329: Contributing guide - uninstalling mne to switch to dev version prone to errors

mmagnuski, 2022-02-13:
Contributing guide advises to install mne via conda:
```
curl --remote-name https://raw.githubusercontent.com/mne-tools/mne-python/main/environment.yml
conda env create --file environment.yml --name mnedev
```
but then, when saying how to switch the python to the dev version, it suggests to uninstall python via pip:
```
pip uninstall -y mne
```
which gave me an error:
```
(mnedev) C:\src>pip uninstall -y mne
Found existing installation: mne 0.24.1
Uninstalling mne-0.24.1:
  Successfully uninstalled mne-0.24.1
ERROR: Exception:
```
```python-traceback
Traceback (most recent call last):
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_internal\cli\base_command.py", line 167, in exc_logging_wrapper
    status = run_func(*args)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_internal\commands\uninstall.py", line 102, in run
    uninstall_pathset.commit()
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_internal\req\req_uninstall.py", line 420, in commit
    self._moved_paths.commit()
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_internal\req\req_uninstall.py", line 273, in commit
    save_dir.cleanup()
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_internal\utils\temp_dir.py", line 173, in cleanup
    rmtree(self._path)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_vendor\tenacity\__init__.py", line 326, in wrapped_f
    return self(f, *args, **kw)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_vendor\tenacity\__init__.py", line 406, in __call__
    do = self.iter(retry_state=retry_state)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_vendor\tenacity\__init__.py", line 362, in iter
    raise retry_exc.reraise()
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_vendor\tenacity\__init__.py", line 195, in reraise
    raise self.last_attempt.result()
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\concurrent\futures\_base.py", line 439, in result
    return self.__get_result()
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\concurrent\futures\_base.py", line 391, in __get_result
    raise self._exception
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_vendor\tenacity\__init__.py", line 409, in __call__
    result = fn(*args, **kwargs)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\site-packages\pip\_internal\utils\misc.py", line 124, in rmtree
    shutil.rmtree(dir, ignore_errors=ignore_errors, onerror=rmtree_errorhandler)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\shutil.py", line 749, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\shutil.py", line 627, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\mmagn\Continuum\anaconda3\envs\mnedev\lib\shutil.py", line 625, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\mmagn\\AppData\\Local\\Temp\\pip-uninstall-67iq3cg9\\mne.exe
```
The permission errors is weird because I run the console as Administrator.
After that I still get mne listed as available, and it seems to have been installed from conda-forge, not pip:
```
(mnedev) C:\src>conda list mne
# packages in environment at C:\Users\mmagn\Continuum\anaconda3\envs\mnedev:
#
# Name                    Version                   Build  Channel
mne                       0.24.1               hd8ed1ab_1    conda-forge
mne-qt-browser            0.1.7              pyhd8ed1ab_0    conda-forge
```

mmagnuski, 2022-02-13:
BTW, after trying `conda uninstall mne` I get mne removed but `mne-base` added...
```
(mnedev) C:\src>conda uninstall mne
Collecting package metadata (repodata.json): done
Solving environment: \
The environment is inconsistent, please check the package plan carefully
The following packages are causing the inconsistency:

  - conda-forge/noarch::mne-qt-browser==0.1.7=pyhd8ed1ab_0
done

## Package Plan ##

  environment location: C:\Users\mmagn\Continuum\anaconda3\envs\mnedev

  removed specs:
    - mne

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    liblapack-3.9.0            |            8_mkl         3.9 MB  conda-forge
    mkl-2020.4                 |     hb70f87d_311       172.4 MB  conda-forge
    numexpr-2.7.3              |   py39h2e25243_2         132 KB  conda-forge
    ------------------------------------------------------------
                                           Total:       176.4 MB

The following NEW packages will be INSTALLED:

  intel-openmp       conda-forge/win-64::intel-openmp-2022.0.0-h57928b3_3663
  mkl                conda-forge/win-64::mkl-2020.4-hb70f87d_311
  mne-base           conda-forge/noarch::mne-base-0.24.1-pyhd8ed1ab_1

The following packages will be REMOVED:

  mne-0.24.1-hd8ed1ab_1

The following packages will be DOWNGRADED:

  libblas                           3.9.0-13_win64_openblas --> 3.9.0-8_mkl
  libcblas                          3.9.0-13_win64_openblas --> 3.9.0-8_mkl
  liblapack                         3.9.0-13_win64_openblas --> 3.9.0-8_mkl
  numexpr                            2.8.0-py39h5ea59f0_101 --> 2.7.3-py39h2e25243_2

Proceed ([y]/n)? y

Downloading and Extracting Packages
numexpr-2.7.3        | 132 KB    | ############################################################################ | 100%
liblapack-3.9.0      | 3.9 MB    | ############################################################################ | 100%
mkl-2020.4           | 172.4 MB  | ############################################################################ | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: done

(mnedev) C:\src>conda list mne
# packages in environment at C:\Users\mmagn\Continuum\anaconda3\envs\mnedev:
#
# Name                    Version                   Build  Channel
mne-base                  0.24.1             pyhd8ed1ab_1    conda-forge
mne-qt-browser            0.1.7              pyhd8ed1ab_0    conda-forge
```

mmagnuski, 2022-02-13:
I don't know what is the difference between `mne` and `mne-base` but I uninstalled it via:
```
conda uninstall mne-base mne-qt-browser
```
and then the installation of dev version went ok.

agramfort, 2022-02-13:
no idea

> Message ID: ***@***.***>
>

mmagnuski, 2022-02-14:
@larsoner If the error is expected then Contributing Guide instructions should be modified, I followed them step by step. :)

PR #10365 (2022-02-23): DOC: Update contrib instructions
Closes #10329

Feel free to merge if you're happy @mmagnuski
closed 2022-02-23 by larsoner