mne-tools/mne-python #10354: unexpected difference in output for `linear_regression_raw` vs. `epochs.average()`

teonbrooks, 2022-02-16:
#### Describe the bug

I noticed that when I use `linear_regression_raw`, this returns a dict of evokeds. However, when I perform the method `epochs.average(by_event_type=True)`, this returns a list of evokeds. I know that there's a major release coming soon so I was wondering if we wanted to keep this pattern or if we wanted to standardize it so that they would both output the same type.

teonbrooks, 2022-02-23:
oh ok, wouldn't we want to change the behavior of `linear_regression_raw` then? It diverges from the list of evoked patterns. no strong feelings about it, just noticed the different output pattern

agramfort, 2022-02-23:
honestly I don't feel any urge changing this but I don't use these very
often myself so...

> Message ID: ***@***.***>
>

teonbrooks, 2022-02-24:
agreed, not that big of a concern. will close

closed 2022-02-24 by teonbrooks