mne-tools/mne-python #10355: Support more header versions for Brain Vision (.eeg) files

FelipeParreira, 2022-02-16:
#### Describe the new feature or enhancement
When I try to read a file with `mne.io.read_raw_brainvision`, I get the following error:
```
ValueError: MNE-Python currently only supports header versions 1.0 and 2.0, got unparsable
\\\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@\x00\x00\x00\x00r\x00\x04\x00\x00\x01\x00\x01\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00mm\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10\x00 \x00\x00\x00\x00\x00\x00\x00\x00\x00?\x00\x00?\x00\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00Created by BESA\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'.
Contact MNE-Python developers for support.
```

#### Describe your proposed implementation
I need the Brain Vision method to be able to read that version of the header file.
If/when you are willing to take a look at this, I can provide the `.vhdr` file I am using with you. 
(Right now, I don't have a file that I can share publicly).

Here is my setup in case it is useful:
```
Ubuntu 18.04.6 LTS
Python 3.8.12
MNE 0.24.1
```

FelipeParreira, 2022-02-28:
HI, @sappelhoff !

Answering your questions:

1. The data was recorded using a device called BW Analysis (from a brand called Neurovirtual). I am not sure about the software itself that was used, though.

2. Here are the files I have. I got this error while trying to read the `.hdr` file (whose extension I changed to `.vhdr`), but I am sending all the files here in case it helps.

[Patient_01.zip](https://github.com/mne-tools/mne-python/files/8153657/Patient_01.zip)

cbrnr, 2022-02-28:
@FelipeParreira I don't think that these are BrainVision files – why do you think that the data is stored in this format? You even renamed the header file, but that doesn't change its contents of course. It looks like this is a completely different format which I have never seen before.

sappelhoff, 2022-02-28:
thanks for the info - that makes the answer to this question very easy: The data you have is **not** in the BrainVision data format, so you cannot read it that way.

Having that said, I don't know the particular data format your files are in, and I recommend you get in touch with the support of the company to learn (i) about the name of the data format used, (ii) how to read these files and work on them, (iii) whether the data can be converted to a more common format such as EDF (https://www.edfplus.info/)

Good luck!

FelipeParreira, 2022-02-28:
I am somewhat new to MNE and eeg files. This just seemed like the last option I had after reading part of the documentation.
If there are other ways to read these files, let me know.

closed 2022-02-28 by cbrnr