mne-tools/mne-python #10360: Only the exact match ch is detected by eog, misc and stim_channel of read_raw_edf

rcmdnk, 2022-02-18:
#### Describe the bug
`eog`, `misc`, and `stim_channel` of [read_raw_edf](https://github.com/mne-tools/mne-python/blob/ad095ba6e0e5b97ddbbb0def3846fa8551add30d/mne/io/edf/edf.py#L1236) are used to assign the channel type for each channel.

Currently, only the exact matched channel can be detected.
Nevertheless, some tutorials, such as [Sleep stage classification from polysomnography (PSG) data - MNE 0.24.1 documentation](https://mne.tools/stable/auto_tutorials/clinical/60_sleep.html#tut-sleep-stage-classif), assume partial match and the results may differ from what is expected.

`stim_channel` is checked by [_check_stim_channel](https://github.com/mne-tools/mne-python/blob/ad095ba6e0e5b97ddbbb0def3846fa8551add30d/mne/io/edf/edf.py#L1156).

`eog` and `misc` is checked directly in[_get_info](https://github.com/mne-tools/mne-python/blob/ad095ba6e0e5b97ddbbb0def3846fa8551add30d/mne/io/edf/edf.py#L440).

I'm not sure if this is a bug or intentional behavior.
The behavior of some programs may change if this is changed to an exact match.
(If the EDF file contains a long name channel whose part matches another stim/misc/eog channel...)

If this is intentional behavior, these tutorials should be fixed.
Otherwise, these codes should be updated to allow the partial match.

#### Steps to reproduce

```python
import mne

files = mne.datasets.sleep_physionet.age.fetch_data(subjects=[0], recording=[1])
raw = mne.io.read_raw_edf(files[0][0], eog=['eog'], stim_channel='marker', misc=['rectal'])
for c, t in zip(raw.ch_names, raw.get_channel_types()):
    print(f'{c}: {t}')
```

#### Expected results

```
EEG Fpz-Cz: eeg
EEG Pz-Oz: eeg
EOG horizontal: eog
Resp oro-nasal: eeg
EMG submental: eeg
Temp rectal: misc
Event marker: stim
```

#### Actual results
```
EEG Fpz-Cz: eeg
EEG Pz-Oz: eeg
EOG horizontal: eeg
Resp oro-nasal: eeg
EMG submental: eeg
Temp rectal: eeg
Event marker: eeg
```

All channels show `eeg` (default type).

#### Additional information
mne.sys_info():

```
Platform:       Linux-5.4.144+-x86_64-with-Ubuntu-18.04-bionic
Python:         3.7.12 (default, Jan 15 2022, 18:48:18)  [GCC 7.5.0]
Executable:     /usr/bin/python3
CPU:            x86_64: 2 cores
Memory:         12.7 GB

mne:            0.24.1
numpy:          1.21.5 {blas=openblas, lapack=openblas}
scipy:          1.4.1
matplotlib:     3.2.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn:        1.0.2
numba:          0.51.2
nibabel:        3.0.2
nilearn:        Not found
dipy:           Not found
cupy:           Not found
pandas:         1.3.5
mayavi:         Not found
pyvista:        Not found
pyvistaqt:      Not found
ipyvtklink:     Not found
vtk:            Not found
PyQt5:          Not found
ipympl:         Not found
mne_qt_browser: Not found
pooch:          v1.6.0
```

cbrnr, 2022-02-18:
I think we should update the tutorial – these parameters are meant to match exact names only. Note that you can always re-assign channels types after reading with [`mne.rename_channels()`](https://mne.tools/stable/generated/mne.rename_channels.html), which is more flexible because it also accepts a callable or a dict.

rcmdnk, 2022-02-18:
Thanks. It's fine for me the usage is confirmed.
Thank you for the rename function, too.

rcmdnk, 2022-02-18:
Yes, I can do at least for the sleep stage classification part 
and will try to find others if exist.

rcmdnk, 2022-02-18:
It seems no other documents have the problem.
(some uses `read_raw_edf(stim_channel=`, but they seem full names of the channels)

agramfort, 2022-02-18:
good catch. It's great if you can send a PR.

> Message ID: ***@***.***>
>

PR #10362 (2022-02-18): BUG: fix stim_channel/misc usage in Sleep stage classification tutorial
#### Reference issue
Fixes #10360.

#### What does this implement/fix?
Fix arguments (stim_channel, misc) of [read_raw_edf](https://mne.tools/stable/generated/mne.io.read_raw_edf.html?highlight=read_raw_edf#mne.io.read_raw_edf) in the [Sleep stage classification](https://mne.tools/stable/auto_tutorials/clinical/60_sleep.html#tut-sleep-stage-classif) tutorial.
closed 2022-02-18 by cbrnr