mne-tools/mne-python #10374: [DOC, BUG] Source Estimate argument vertices wrong type

alexrockhill, 2022-02-22:
The Volume Source Estimates (vector and plain) both say in the docstring that they should take an array of vertices (https://mne.tools/dev/generated/mne.VolVectorSourceEstimate.html#mne.VolVectorSourceEstimate and https://mne.tools/dev/generated/mne.VolSourceEstimate.html#mne.VolSourceEstimate). When this is followed a cryptic error from `np.diff` is thrown. SourceEstimate (https://mne.tools/dev/generated/mne.SourceEstimate.html#mne.SourceEstimate) has a nice explanation that they are supposed to be lists of length 2 for each hemisphere. Since the volume combines the hemispheres, this could be a list of length 1 but that's a bit unnecessary, it seems that the documentation maybe shouldn't be changed but the vertices should be wrapped in a list when passed out of the volume source estimates.

alexrockhill, 2022-02-22:
Ah, looking into it, because of the mixins, it's maybe not trivial to pass it as a list, maybe it should just be a list of length 1 in the docstring....

larsoner, 2022-02-22:
Some time ago we tried to change everything so that `vertices` should always be passed as a list. A list of length 1 is fine for VolSourceEstimate (though it can be more if you use subvolumes!). Let's just update the docstring to say it should be a list of ndarray

closed 2022-02-22 by larsoner